kiteretro/Circuit-Sword #5: Digital Volume Controller - Button Remapping

markolomew, 2018-02-11:
@kiteretro
Please document how to alter the board so that we can reassign the Digital Volume Controller to L1/R1 or L2/R2 instead of buttons 5 and 6.

kiteretro, 2018-02-11:
Fixed, see guide here: https://github.com/kiteretro/Circuit-Sword/wiki/Digital-Volume-Guide#alternative-pins-for-digital-volume-control

Note that for the NEXT CSO order I have changed this so it's a solder jumper, so cutting needed :)

closed 2018-02-11 by kiteretro

markolomew, 2018-02-11:
> Fixed, see guide here: https://github.com/kiteretro/
> Circuit-Sword/wiki/Digital-Volume-Guide#alternative-pins-
> for-digital-volume-control
>
> Note that for the NEXT CSO order I have changed this so it's a solder
> jumper, so cutting needed :)

Is the best way to solder this volume control switch down to pre solder all
the points on the CS then lay the control switch on top, apply a little
downward pressure on the switch,  and heat up each solder point one by one?
Just wondering as I haven't seen this part in practice soldered yet.
Mark

kiteretro, 2018-02-11:
@markolomew yep, I would do exactly that however with a slight bit of clarity: pre-tin all the pads on the PCB itself, but make sure to not have very much solder on there. Too much solder may make soldering it not very level, but basically what you say is what I'd do :)