kiteretro/Circuit-Sword #15: Crash and Joystick

TigerClaw33, 2018-02-17:
Hi,
I have two problems. At first sometimes retropie crash and I have to turn the power of and push the "mode" button to turn the device off. 

and second I calibrated the analog joystick and configure all buttons in retropie. But if I start an N64 game like mario cart the joystick didn't work.

kiteretro, 2018-02-17:
Can you disable wifi with the MODE + B button combo and let me know if the crashes continue? Is it in game or in ES menu? Does the sound also freeze? Can you unplug and re-plug the CM3? Are you powering it with a battery?

As for joystick, if you follow the use case here: https://github.com/kiteretro/Circuit-Sword/wiki/Configuration-Switches for "Testing the USB devices".. and when you plug in to PC and turn on you should see some USB devices appear, and if you open "set up USB game controllers" from the start menu, you can verify the joystick inputs.

If all is well, you may well want to try a fresh image or 'set up game controllers' from the ES menu. Making sure that JOY1 is the joystick you configured.

TigerClaw33, 2018-02-18:
- never happend again with crash
- WIFI is not working. I can see my box but if I set the password nothing happend. Cannot connect to network
- Kombo MODE+B not working, sometimes it work sometimes not. At the moment it works with button A
- Kombo with digital audio control works at the beginning. now it's not working
- Joystick works with other emulator, but up/down and left/right is down/up and right/left. If I configure the buttons in the menu the other way around it's the same up is down and down is up.

kiteretro, 2018-02-19:
- Wifi - try being physically closer to AP, the wifi isn't very strong
- Update arduino and run ./update.sh script (see wiki for both) i have pushed some updates
- While you are uploading the arduino code, edit the joystick direction at the BOTTOM of config.h (in arduino code)

TigerClaw33, 2018-02-20:
Device was very closed to gameboy but no connection possible. USB C to PC also not possible. No driver found. Because I tried to updated retropie via USB ...

TigerClaw33, 2018-02-20:
How to update the code? don't find the right code part?

kiteretro, 2018-02-28:
It's on the wiki https://github.com/kiteretro/Circuit-Sword/wiki/Joystick-Guide

Closing this issue as joystick is working and no further reports

closed 2018-02-28 by kiteretro