kiteretro/Circuit-Sword #16: Create new combo key to turn on/off joystick

markolomew, 2018-02-18:
I'd like to request a new combo key to enable/disable joystick output.

kiteretro, 2018-02-18:
Feature ask is to add a button combo that when activated will always report the joysticks at idle position even if they are moved. This is in the case where user does not want joystick input in places where knocking the sticks is annoying and affects use.

timetraveler0110, 2018-02-20:
I don’t understand a word
Half because I don’t speak English 
Other half because I don’t have computer knowledge
Can anybody explain this as if I were an Spanish child?

kiteretro, 2018-02-21:
@timetraveler0110 Those are my notes to remind me for when I implement it, there is nothing for you to do in this issue as it is not solved.

markolomew, 2018-04-19:
Now that I've been playing around with the joystick in the bottom corner setup, I'm not really noticing any issues of accidentally hitting it like I had first though.  Unless anyone else sees this as still a worthy issue, I would say you can close this feature request.

kiteretro, 2018-04-20:
Thanks, I'll leave it open for now as my 'redo' of the software will have the option to enable/disable them without needing to code, so it would solve this as well as making the first time setup easier too :)