kiteretro/Circuit-Sword #17: Volume combo with mode + up/ down doesn’t work

timetraveler0110, 2018-02-19:
When I try to turn up or down the volume with mode button + up/ down, it doesn’t work.
Brightness combo works although.

TigerClaw33, 2018-02-19:
exactly the same error :(

kiteretro, 2018-02-19:
Could you both also update the arduino code, and also run the ./update.sh (in the Circuit-Sword directory on the Pi. I have pushed some changes that may help!

timetraveler0110, 2018-02-20:
I’m sorry Kite, I connected Swords USB-C cable to PC USB-A and PC doesn’t recognise Arduino Leonardo on GBoy. I put the switches B and C on. 
As Circuit Sword has 2 USBs the USB-C and the uSB-A. Maybe I should have connected the Swords USB-A , but I don’t have now a USB-A male to male cable.
Am I right or wrong?
Why PC doesn’t recognise Sword , nor USB Audio, nor Arduino Leonardo?
Thanks Kite, and sorry for my ignorance.

timetraveler0110, 2018-02-20:
I connected via SSH , updated executing update.sh, and did what you told me 👍 
I did:
SSH in
sudo service cs-osd stop
miniterm.py /dev/ttyACM0
Press the "f" key, each press should show you a number, this is what the arduino is telling the Pi to set the volume to.
Hold MODE + UP/DOWN for a few seconds
Press the "f" key again, does the number change?
CTRL + ] (to exit)
sudo service cs-osd start
——————
And the value was always 255
The value didn’t change

timetraveler0110, 2018-02-20:
I have tried on Ubuntu 16.04 AMD64 to update Arduino code but the same, not recognised...
First try was on Win 7 AMD64...

jermzz1, 2018-02-20:
I updated the Audrino code and updated via the update.sh and the problem is the same. I responded this in my thread on Sudomod, but I saw you prefer here, so I will post it here, too. I fixed my inverted axis on my joystick, so I know the code took when I uploaded it. still getting 255 when I take the audrino reading.

kiteretro, 2018-02-20:
the `./update.sh` does NOT update the Arduino code. The issue is on the Arduino side.

RE: Not being detected on Windows, do you have a USB hub somewhere, like ANY USB hub? Plug the HUB into the PC, and then the CSO into the hub.. you also need to turn the power switch ON on the CSO.

Alternativly! I have written a tool that will update the arduino from the Pi itself! Can you run the `./update.sh` script again, and then you should also get `./flash-arduino.sh` :) can you please try running that. You will probably need to run `sudo apt-get install avrdude` first, and then `sudo ./flash-arduino.sh`

jermzz1, 2018-02-20:
Kite, although I am sure I updated the audrino manually correctly, I tried your new method. Kudos on that integration! It works well!

Although as with the manual update. The problem has not changed. I still have no volume control.

kiteretro, 2018-02-20:
Ok I think I'm getting closer to what the issue is! I have just pushed another change, can you please run `./update.sh` again and also `./flash-arduino.sh` (note that you will need to have rebooted since the last time you ran it).

Once done (and rebooted again) follow the instructions for opening `miniterm.py` but instead press `J` once. This will start the joystick calibration and when done it should say `OK` afterwards. Now with the window still open press the `f` key again, any change?

If you are still not having change, please download the arduino code again to PC and in `config.h` at like 67, remove `#define USE_VOLUME_ANALOG` completely.. Re-upload and calibrate joystick (either from the startup button, or using miniterm as above. Please let me know the results :)

timetraveler0110, 2018-02-20:
i tried ./flash-arduino.sh, and got this
[image]

kiteretro, 2018-02-20:
Don't call it with sh, you should instead run as `sudo ./update-arduino.sh` .. it does make a difference!

timetraveler0110, 2018-02-20:
hi Kite
i did it all as it follows
Ok I think I'm getting closer to what the issue is! I have just pushed another change, can you please run ./update.sh again and also ./flash-arduino.sh (note that you will need to have rebooted since the last time you ran it).

Once done (and rebooted again) follow the instructions for opening miniterm.py but instead press J once (it must be a capital "J", so SHIFT + J). This will start the joystick calibration and when done it should say OK afterwards. Now with the window still open press the f key again, any change? Make sure to do the MODE + UP/DOWN a couple of times and f a few more times afterwards.

If you are still not having change, please download the arduino code again to PC and in config.h at like 67, remove #define USE_VOLUME_ANALOG completely.. Re-upload and calibrate joystick (either from the startup button, or using miniterm as above. Please let me know the results :)

------------------------------------------
it changed values from 100 to 80, 45, 35, an so on, so , IT CHANGES VALUES NOW , with shift+J
I didnt have to remove  #define USE_VOLUME_ANALOG

timetraveler0110, 2018-02-20:
i started Retropie now and IT SEEMS TO WORK NOW!
THE VOLUME COMBO WORKS NOW!

jermzz1, 2018-02-20:
Kite, 

Volume seems to be working as expected now. It is a bit slow to update, but I won’t gripe about this as you said you’re going to be improving it all around. 

Flashing the audrino from the pi looks like it takes directly from your repository, is that correct? Only asking because my joystick is inverted again. 

I should just be able to download the newest version and flash via audrino IDE after inverting my joystick, correct? Will I need to run calibration again via miniterm.pi? It wasn’t til then did my volume start working. 

Thanks again for quick problem resolution. Top notch customer service :)

kiteretro, 2018-02-20:
@jermzz1 Yep correct it is a pre-compiled version that i make (see the .hex in the arduino source location) which is all defaults. Flash now with your changes and all should be good. You shouldn't need to re-calibrate :) 

I have an issue already for the slow update, or at least I'll make a new one soon.. i do plan to address that soon as it's a bit annoying the slow update.. but at least this bug is now fixed! Thanks all for your help!

I will also make another script that will do a compile of the code and upload from the pi, so that if you edited the files on the pi and did the upload they would take effect (rather than currently using the .hex that i built all the time).

I'm going to close this issue now :) I'll open some new ones soon to track making the volume more responsive!

closed 2018-02-20 by kiteretro

TigerClaw33, 2018-02-20:
I did update with sudo ./update.sh but volume not working.

kiteretro, 2018-02-20:
@TigerClaw33 you must also update the arduino code, and then re-run the calibrate joystick.. let me write an easy to understand page :)