kiteretro/Circuit-Sword #21: Add Volume OSD overlay

c-hri-s, 2018-02-21:
It would be useful to have a visual representation of the current volume on-screen when you're changing it, so you can understand how loud you're adjusting it. Without a physical volume switch it's difficult to gauge.

This would be particularly useful when modifying it from ES when you may be adjusting it with no sound actually playing.

Something like a simpler version of the iPhone volume graphic https://i.stack.imgur.com/OJmvI.gif

kiteretro, 2018-02-21:
Agree 100% and I have ideas on doing so :) It will also do the same for brightness, and any other 'state change' (e.g. turning wifi on/off).

It will also incorporate a menu that will be able to adjust certain settings (e.g. 'calibrate joystick now', 'calibrate battery voltage', 'invert joystick axis', etc)

Thanks for raising this, I will be using it to track progress.

c-hri-s, 2018-02-21:
Sounds great!

StanMOI, 2018-04-14:
waiting for update

StanMOI, 2018-04-15:
Did anyone already done this option by him self?

kiteretro, 2018-04-16:
I have some work in progress, nothing committed yet, but don't worry i'm still working on this :)

StanMOI, 2018-04-18:
YES!
You are the BEST !

cvondra, 2018-07-31:
Any ETA on when we may see this functionality?

kiteretro, 2018-08-01:
Within the next 2 months. The software that contains this is also required by the Circuit Sword Lite, so when those are here (they were in the latest order a month ago) I literally HAVE to have it ready, and I'll do it for the CSO too :) there are some other notable improvements so it'll be worth the wait!

kiteretro, 2018-10-03:
Fixed in https://github.com/kiteretro/Circuit-Sword/releases/tag/v1.3.0

closed 2018-10-03 by kiteretro

c-hri-s, 2018-10-03:
Certainly is - nice OSD, thanks Kite! :)