kiteretro/Circuit-Sword #22: Crash when Wi-Fi is turned on

sjambon, 2018-02-21:
When I have my wifi turned on, my retropie freezes before I can complete configuring my controller. After my CS freezes, I have to turn it off by pressing the mode button for a second. When my wifi is turned off, I don't have this issue. Is there a possibility to fix this issue?

kiteretro, 2018-02-21:
Can you please put SD in PC and edit (with Notepad++, NOT the regular notepad) `config-cso.txt` and change the last line so that is exactly like this:

``` bash
# Startup Exec
STARTUPEXEC="sudo cp /var/log/messages /boot/messages.txt"
```
Then put back in the CSO and boot up fully .. wait until a crash .. power it off.. boot it again .. power it off (basically to get 2x reboot cycles)
Then put in PC and upload the new file `messages.txt` here.

You can also stop WiFi from loading COMPLETELY by editing `config.txt` and putting a `#` in front of the line, so that it looks like the following:

``` bash
#dtoverlay=sdio,poll_once=false
```

sjambon, 2018-02-21:
[messages.txt](https://github.com/kiteretro/Circuit-Sword/files/1745633/messages.txt)
[messages1.txt](https://github.com/kiteretro/Circuit-Sword/files/1745634/messages1.txt)
Here are the 2 files you requested.

kiteretro, 2018-02-21:
Is your CSO fully built and in a case? If it is can you take the two halves of the case apart, and boot? There is already a known issue where the wifi gets shorted https://github.com/kiteretro/Circuit-Sword/wiki/Known-Issues#issue-4---wifi-doesnt-work so can you check that that isn't the case first? Can you also try re-imaging (again) now that the case in in two halves (to make sure the crashes haven't corrupted anything).

Secondly, once you have tried that, can you try the 2nd thing that I mentioned about disabling wifi in the `config.txt` ? Again re-image (and edit the config.txt file) to make sure it's not being corrupted.

From the logs, bad stuff happens around wifi INIT. For example:

```
Jul  5 11:46:49 retropie kernel: [   72.403276] RTL8723BS: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Jul  5 11:46:49 retropie kernel: [   72.403358] RTL8723BS: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Jul  5 11:46:50 retropie kernel: [   72.632288] RTL8723BS: rtw_cmd_thread: DriverStopped(0) SurpriseRemoved(1) break at line 458
Jul  5 11:46:50 retropie kernel: [   72.671655] RTL8723BS: rtw_ndev_uninit(wlan0)
Jul  5 11:46:50 retropie kernel: [   72.822526] RTL8723BS: rtw_dev_unload: driver not in IPS
Jul  5 11:46:50 retropie kernel: [   72.823128] mmc1: card 0001 removed
Jul  5 11:46:53 retropie kernel: [   75.750754] mmc1: new high speed SDIO card at address 0001
Jul  5 11:46:53 retropie kernel: [   75.751470] pnetdev = a359a800
Jul  5 11:46:53 retropie kernel: [   75.827834] RTL8723BS: rtw_ndev_init(wlan0)
Jul  5 11:47:14 retropie kernel: [   96.931518] NMI backtrace for cpu 0
Jul  5 11:47:14 retropie kernel: [   96.931525] CPU: 0 PID: 940 Comm: wpa_supplicant Tainted: G         C      4.14.14-v7+ #1
```

If the two things at the top of this message do nothing, then I'll need to replace your board. Please send me an email (once those other things have been tested) and we'll sort out the return :)

sjambon, 2018-02-23:
Disabling the Wi-Fi fixed the crashes but which chip on the Circuit Sword is the Wi-Fi chip so I can put some capton tape on it?

kiteretro, 2018-02-23:
See here: https://github.com/kiteretro/Circuit-Sword/wiki/Circuit-Sword-Original-V1.1E#bottom

Note that it sounds like there is something either broken with the wifi chip or there is solder loose on the chip/board. Most likely to repair will require some soldering.. if you would like me to repair (free of charge) then please send me an email and we can work out the return.

If you are ok with USB wifi then the current solution works. But it's up to you.

sjambon, 2018-02-23:
Where can I find your E-mail adress so I can contact you?

kiteretro, 2018-02-23:
kite@kitesitemshop.com

kiteretro, 2018-02-28:
Issue most likely WiFi related, either faulty chip or inconsistent connection (maybe a break in soldering). Waiting on board to verify, closing the issue for now.

closed 2018-02-28 by kiteretro