kiteretro/Circuit-Sword #23: Crash on first boot

jamesalmeida, 2018-02-23:
I put the premade image on my SD card and booted up the Circuit-Sword. Thought it was taking a while but I just let it go for a bit just in case it was still working. The screen finally updated but then I got this error and it hasn't changed now for about 15 minutes. What does this error mean?

[image]

kiteretro, 2018-02-23:
Hi James,

Looks like wifi causing problems.. are the two halves together? If they are can you take the back off (in case it is the back board pressing into the WiFi module) and try re-booting..

We can stop wifi loading COMPLETELY by editing config.txt and putting a # in front of the line, so that it looks like the following:
``` bash
#dtoverlay=sdio,poll_once=false
```

That should get it booting.. once booted can you press F4 on a keyboard and do `sudo cp /var/log/messages /boot/messages.txt` and then put back into PC and upload the `messages.txt` file that should be there now

jamesalmeida, 2018-02-23:
First thing I tried was disabling the WIFI like you said by editing config.txt and that seems to have fixed it. It boots up and works now. I'll attached my messages.txt below. 

[messages.txt](https://github.com/kiteretro/Circuit-Sword/files/1752657/messages.txt)

kiteretro, 2018-02-28:
Ok James, looks like there is some hardware fault with the wifi module or the soldering of the module. If you would like to send it to me I can repair (free of charge). Sorry for the problems! Send me an email at kite@kitesitemshop.com and we can go from there.

closed 2018-02-28 by kiteretro