kiteretro/Circuit-Sword #24: [Issue] No sound from speaker

jamesalmeida, 2018-02-23:
Sound works through the headphone jack but not through the speaker. I soldered the red wire to the + and black wire to the - on the speaker and then plugged it into the speaker port on the board. Is that how it's supposed to be installed?

kiteretro, 2018-02-23:
That all sounds good James, could you see if in the top right corner there is a speaker icon shown? If yes then that's the MUTE icon (which disables the amp) .. could you do like MODE + X (i think) to disable it, otherwise it's MODE + Y ?

Send a pic of how it's connected, make sure the connector is all the way in :)

kiteretro, 2018-03-01:
Did you get this fixed James?

jamesalmeida, 2018-03-02:
Haven't been able to check connector and take a pic of inside yet.

jamesalmeida, 2018-03-04:
The connector is all the way in and the sound is turned on. Still nothing comes out of speaker. Only sound that works is through the headphone jack.  

Here's a pic of my speaker:
[image]

kiteretro, 2018-03-04:
Have you got another speaker to try? Like on those solder pads next to it? 

Could you measure the voltage on this resistor to gnd: 

[image]

kiteretro, 2018-07-01:
Closing issue, re-open or send me an email if needed.

jamesalmeida, 2019-01-08:
I measured it and it says 3.27

jamesalmeida, 2019-01-08:
@kiteretro I measured it and it says 3.27

Still no sound from speaker. Tried another speaker too. I do get sound through headphone jack though.

kiteretro, 2019-01-08:
From that picture above, it looks as if the speaker trace has been damaged (see just above where the "S-" hole connector is. Could you try and clean up that area and get a clear photo of it. It might help to have some strong light and view it from an angle..

What you might need to do is gently scrape the black off the trace and bridge it with some solder.

This is what that area looks like from the design:

[image]

jamesalmeida, 2019-01-08:
@kiteretro That stuff above the S- hole is glue from my hot glue gun that was holding the speaker in place. Do you think hot glue could have damaged the trace?

jamesalmeida, 2019-01-09:
@kiteretro Sound just started working! Thanks for your help!

closed 2019-01-09 by jamesalmeida