kiteretro/Circuit-Sword #26: Update should update not reinstall

c-hri-s, 2018-02-27:
Hi,

Your update script takes a fairly brutal approach to upgrading, overwriting /boot/config.txt and so resetting overscan settings, and overwriting ES configuration settings so losing favourites, last played, skin, etc.

Is there a way to have it upgrade but leave modified files so user configuration/changes aren't lost?

Thanks

kiteretro, 2018-02-27:
Agreed.. so the following should probably happen:

- Don't copy config.txt if it already has DPI (or some searchable) settings already applied (I see in my code there is a bug here, it does check it correctly BUT it still copies the files.. https://github.com/kiteretro/Circuit-Sword/blob/master/install.sh#L111)
- Don't copy ES settings if changes applied https://github.com/kiteretro/Circuit-Sword/blob/master/install.sh#L156

I will probably add a flag or some file that tells the script "this is an update" because it is currently designed that `install.sh` is called and it has very basic settings applied..

For the time being you can do a git pull in the directory, and then edit those lines above to remove the changes you don't want and then run `./install.sh YES` .. it is only the `update.sh` script that does the pull and then auto-install which you want to avoid for now..

c-hri-s, 2018-02-27:
Sounds good. Thanks for the workaround in the meantime.

c-hri-s, 2018-02-27:
You might like to consider rxbrad/gbz35 as the default theme instead of pixel. Might be a case of personal preference, but I find it much clearer on a small screen.

kiteretro, 2018-02-27:
Thanks, I can pre-install it as part of the script and I'll try it out.. i've always used Pixel personally, but willing to give a go :) https://github.com/rxbrad/es-theme-gbz35

jarettmillard, 2018-03-30:
There's also Pixel TFT for small screens: https://github.com/djbriane/es-theme-pixel-tft