kiteretro/Circuit-Sword #27: [issue] I can not booted up the system

FoRTu, 2018-02-27:
Hello,

I don't know if I'm doing something wrong but I can not booted up the system. I try diferent types of SD cards getting the same result, the screen flash after I switch on but nothing apears on the screen. As I can read on FAQs I wait some minutes to resize the file system but aften 10 minutes the board looks the same.

I try to [disable the wifi](https://github.com/kiteretro/Circuit-Sword/wiki/Troubleshooting-(LCD-Power-USB-Software)#disabling-wifi-completely) editing the config.txt but still the same.

Here is a image of how looks the board when is booted up:

[image]

You can see on the image the two types of SD card I use (with the same result).

Thank you

kiteretro, 2018-02-27:
Hm, the 1V8 LED isn't on, if you remove the CM3 and power the board does the 1v8 LED come on at all? If it doesn't, can you send me an email as it'll need to be returned as it is faulty and I will send you out a new one right away! So please send me an email to kite@kitesitemshop.com :)

FoRTu, 2018-02-28:
Thank you for your quick response.

Yes, the 1V8 LED isn't on as you can see in the image below, with and without CM3. 

[image]

And Here is the picture of the underneath of the CM3:

[image]

Before send you a mail to proceed to return the board, is there any solution to solve it? soldering something?

Thank you.

kiteretro, 2018-02-28:
The only thing that looks amis is the inductor here:

[image]

Do you have hot air reflow equipment? Or do you have solder and flux? If possible, try adding hot solder to either side of that inductor, one at a time..

FoRTu, 2018-02-28:
I don't have any hot air reflow equipment, and I think that haven't enough experience soldering to do that. So, I think the best way is to return the board and replace for another one.

I send you a mail to kite@kitesitemshop.com to proceed.

Thanks for the support.

kiteretro, 2018-03-19:
Closing as board has been repaired and will be sent back shortly

closed 2018-03-19 by kiteretro