kiteretro/Circuit-Sword #28: [Next] One time reboot to HDMI with full resolution

kiteretro, 2018-03-01:
Some kind of menu option for "One time reboot to HDMI" which will allow for HDMI to be FULL HD resolution, and then cloned back to the internal screen.. this will allow you to actually use it on a TV with something nicer than 320x240!

jfenton78, 2018-07-06:
A recent post on reddit discuses auto-switching between an LCD and HDMI-Out. This may be a solution for the one time reboot to HDMI.

https://www.reddit.com/r/raspberry_pi/comments/6lq09q/autoswitch_between_lcd_and_hdmiout/

kiteretro, 2018-07-06:
Indeed, the complication is linking it up to the `dpi-cloner` binary because different parameters would be needed to clone from HDMI to the internal screen. It's all very possible :) low on my list because HDMI and LCD cloning already works, but def can do with improving as you have suggested. Thank you for the link!

kiteretro, 2018-09-22:
+ add audio over HDMI

kiteretro, 2018-10-03:
Fixed in https://github.com/kiteretro/Circuit-Sword/releases/tag/v1.3.0

closed 2018-10-03 by kiteretro