kiteretro/Circuit-Sword #29: Wifi isn't working correctly

jamesalmeida, 2018-03-01:
My Wifi wasn't working and you asked me to open the case to make sure it wasn't a short. Interestingly, the wifi started working when I had the case open. When I closed it up very carefully to make sure there was no short or anything the Wifi doesn't work again. Any idea what's going on? It just says on my screen that there's no Wifi module available when I try to connect to Wifi. I use the button combo to make sure it's on the top of the screen and at first it shows with an exclamation point (!) but then after a few seconds changes to having an X across the Wifi symbol.

kiteretro, 2018-03-01:
Have you covered the wifi module with kapton tape? Is anything putting pressure on the fan/CM3 when you close the case? Like wires for the power switch or the back plastic of cart?

jamesalmeida, 2018-03-01:
I haven't covered the wifi in kapton tape. I don't have a the fan on the CM3 yet so there's plenty of space above the CM3 and there's no cartridge at all so that part is just open.

DaveyD71, 2018-03-01:
I had the exact same problem with mine. Open the case all works, screw it up WiFi goes (!) then (X). I found that the back board is pressing against the WiFi part of the front board. So I put a peace of foam tape on the back board where the two would touch so creating a gap between the two and all works great. Hope this helps.

kiteretro, 2018-03-01:
@jamesalmeida Yeah ok add kapton tape or foam to the wifi module, there is a known issue for this: https://github.com/kiteretro/Circuit-Sword/wiki/Known-Issues#issue-4---wifi-doesnt-work

@DaveyD71 foam works too :) also trimming those USB legs a bit will help

jamesalmeida, 2018-03-01:
Thanks @kiteretro and @DaveyD71 

I'll definitely give this a try!

jamesalmeida, 2018-03-04:
I trimmed the USB legs and also covered it with electrical tape and also the wifi chip is now covered with a piece of electrical tape. Works now! Thanks!

closed 2018-03-04 by jamesalmeida