kiteretro/Circuit-Sword #31: update.sh fails due to missing "/etc/init.d/asplashscreen" file

lhess, 2018-03-11:
I am using your provided retropie image for my **CSO** and updated retropie by it's own script (Retropie-Setup/retropie_setup.sh) to version **4.3.13**.

If I try to execute your update.sh script now, it fails with:
`sed: can't read /etc/init.d/asplashscreen: No such file or directory`

Here the complete output:
```
pi@retropie:~/Circuit-Sword $ sudo ./update.sh
Stopping OSD..
Stopping CLONER..
Updating OSD from git..
Fetching origin
HEAD is now at eebe401 Update README.md
Performing script setup..
INSTALLING..
[i] FILE: [/home/pi/Circuit-Sword/LICENSE] exists.
[*] EXECUTE: [chown -R pi:pi /home/pi/Circuit-Sword]
[i] FILE: [/boot/config_ORIGINAL.txt] exists.
[*] EXECUTE: [cp /home/pi/Circuit-Sword/settings/config.txt /boot/config.txt]
[*] EXECUTE: [cp /home/pi/Circuit-Sword/settings/config-cs.txt /boot/config-cs.txt]
[*] EXECUTE: [cp /home/pi/Circuit-Sword/settings/asound.conf /etc/asound.conf]
[i] FILE: [/opt/retropie/configs/all/autostart_ORIGINAL.sh] exists.
[*] EXECUTE: [cp /home/pi/Circuit-Sword/settings/autostart.sh /opt/retropie/configs/all/autostart.sh]
[*] EXECUTE: [chown pi:pi /opt/retropie/configs/all/autostart.sh]
[*] EXECUTE: [chmod +x /home/pi/Circuit-Sword/cs-osd/cs-osd/cs-osd]
[*] EXECUTE: [chmod +x /home/pi/Circuit-Sword/rfkill/rfkill]
[*] EXECUTE: [chmod +x /home/pi/Circuit-Sword/dpi-cloner/dpi-cloner]
[*] EXECUTE: [chmod +x /home/pi/Circuit-Sword/cs-tester/pngview]
[*] EXECUTE: [chmod +x /home/pi/Circuit-Sword/install.sh]
[*] EXECUTE: [chmod +x /home/pi/Circuit-Sword/update.sh]
[*] EXECUTE: [chmod +x /home/pi/Circuit-Sword/flash-arduino.sh]
[*] EXECUTE: [sed -i "s/ *both/ alsa/" /etc/init.d/asplashscreen]
sed: can't read /etc/init.d/asplashscreen: No such file or directory
ERROR: Command exited with [2]
Starting OSD..
DONE!
```

kiteretro, 2018-03-11:
Should be fixed now

lhess, 2018-03-12:
Yes, it works!

kiteretro, 2018-03-12:
Hooray! :) Closing

closed 2018-03-12 by kiteretro