kiteretro/Circuit-Sword #36: Select and Start not Mapping correctly

thebadbishop, 2018-03-20:
Hi, my unit is saying the select button is already taken when I attempt to map it.
I skipped the mapping of select, and it turns out the select and start buttons are both regestering as a “Start” push.

Please let me know how to fix this issue.

kiteretro, 2018-03-21:
Hi, can you please take a close up, in focus, in light, picture of this area (it is just above where the DPAD is):

[image]

Could you also take a picture in the same way of the start and select buttons:

[image]

Are you also able to connect it to a PC and use the "USB GAMEPAD" option in windows to see what button does what? https://github.com/kiteretro/Circuit-Sword/wiki/Configuration-Switches

thebadbishop, 2018-03-21:
No PC access at this moment, but here are some photos:
[image]
[image]
[image]
[image]
[image]

thebadbishop, 2018-03-21:
Got a little closer on what I think is the issue (after looking at the 2 places you pointed out)
[image]

kiteretro, 2018-03-21:
There's your issue :) how are you with a soldering iron and sucker/braid? Do you know anyone who may be able to help?

What you need to do is apply flux to the area (this is a MUST) and then heat up the solder bridge, then using a solder sucker on it to pull away the excess!

I can repair this for free if you send me an email, but it would be easier if someone local could help you. It's worth trying that first, and as long as you literally don't put HOLES on the board then whatever the result is I can repair anyway..

thebadbishop, 2018-03-24:
That worked!

... but now my unit is flickering... the screen flickers off an on and it seems to be running hot not sure if related (and i have both the fan and heatsink working).

kiteretro, 2018-03-28:
Ok is this still happening? If you remove the CM3 and under where it is, should be an LED "BL", is this glowing brightly? ALso is the chip next to it really hot? (be careful it may burn you if so)? Or are you meaning some other chip is hot?

thebadbishop, 2018-03-28:
The flickering seems to only be an issue when the battery is low.

As for the heating, it doesn’t seem to be as bad now (I removed my cosmetic cartridge to allow for more airflow)

kiteretro, 2018-03-28:
Ok, I'm going to close this issue as fixed. If there are problems with any component getting too hot, or anything stops working, please let me know (e.g. raise a new issue or email me). In normal use nothing should get too hot for you to hold your finger on.

closed 2018-03-28 by kiteretro