kiteretro/Circuit-Sword #39: Volume Weel (volume rocker) doesn't work

StanMOI, 2018-04-14:
Hi Kite,
I follow the guide for digital volume and soldered the volume https://www.aliexpress.com/item/10pcs-l ... 57524.html
<img>https://github.com/kiteretro/Circuit-Sword/wiki/images/CSO/CSO_DIGITAL_AUDIO_2.PNG</img>
Changed the two lines to:
```
#define USE_VOLUME_DIGITAL
#define USE_ALT_PINS_VOLUME_DIGITAL
```
But my weel doesn't work.
Only Mode button with volume fonction makes volume change.

Please help

StanMOI, 2018-04-14:
Thanks.
I uncomment all three lines in "config.h" 
```
#define Use_Old_Button_Combos
#define Use_Volume_Digital
#define Use_Alt_Pins_Volume_Digital (if using 6 button front)
```
than upload it via SFTP using "Cyberduck".

Still nothing.
I will try to test all with Multimeter to be sure
there aren't any shorts on the switch.

Maybe i make some mistake?
When you say _"upload to arduino"_ you mean that i have to use Arduino software to do it?

markolomew, 2018-04-14:
You can't use cyberduck to update the arduino, sadly. You have to go to
arduino's website and download the arduino software to do it. It's free.

Look on this github page for step by step instructions on how to do it with
the circuit sword:
https://github.com/kiteretro/Circuit-Sword/wiki/Updating-Arduino-(button-controller)-Firmware

Mark

StanMOI, 2018-04-14:
OK, thank you.
I will try to do it.
But look at that  _"It is also possible to update the arduino from the Pi itself."_
Enable WiFi and SSH
SSH in
```
cd Circuit-Sword
sudo apt-get install avrdude
sudo ./update.sh YES
sudo ./flash-arduino.sh
sudo reboot
```
https://github.com/kiteretro/Circuit-Sword/wiki/Updating-Arduino-(button-controller)-Firmware

StanMOI, 2018-04-14:
I done it with arduino software.
know my wheel work at half...
So when i turn wheel to :
DOWN VOLUME = work's, but not every time.
UP VOLUME  doesn't work at all.

Maybe i made some shorts when solder it.
:(

markolomew, 2018-04-14:
It's possible but it didn't work for me.

StanMOI, 2018-04-14:
You mean that you have the same problem with Wheel ?

markolomew, 2018-04-14:
My digital volume control switch (rocker switch) works properly. Update the
arduino using the arduino software, double check your solder points on your
rocker switch, and you should be good to go.

StanMOI, 2018-04-15:
"**check your solder points on your
rocker switch**"

Where exactly i have to check , between which point?
On "rocker switch", there is 3 points: 1, 2, T
On PCB where i have to touch with multimeter to test one of this points (1,2,T)?

kiteretro, 2018-04-16:
Maybe already covered, but have you done a joystick calibration at all using the board? In order to get analog volume working easily I made it so that joystick calibration did stuff relating to it.. if you have done a calib with old code there is a chance it's messing up, so doing a re-calib on newer code may help?

StanMOI, 2018-04-16:
Hi, Kite,
Thank you for support.
Why you say _"...analog volume working"_, isn't a "rocker switch" not digital one?
I never done the calibration of joystick. I always used EMULATIONSTASTION INPUT Configure to configure my joystick and it's always worked.
What's is the right order : 
first -> calibration of joystick, 
second -> change 2 lines in "config.h" 
or doesn't matter ?

In the case , to be sure the i all done..
Where exactly i have to check with multimeter, between which point on **PCB** and 3 points of "rocker switch: **1, 2, T**?

kiteretro, 2018-04-16:
To clarify as I may be making assumptions in understanding:

- CSO volume control is digitally done via the Pi
- There are a number of ways to actually adjust the volume:
  - Via button combos (always works)
  - Via analog volume where (defined as ANALOG volume control)
  - Via the C1/C2 or rocker switch (defined as DIGITAL volume control)

All methods are linked, and I should have implemented it better in the first place (software update coming soon to help here).. Does the volume ever do something like jump to 100% or 0% or something? If so, then you need to do a "joystick calibration" by holding START when powering on the CSO (even if you have no physical sticks).

You should be able to use a multimeter on the pins you have listed, and they should go to 0v (or short to GND if in resistance mode) when switched. One probe needs to be on GND, and the other on the pins. You will need to make sure that the GND pin is soldered properly, which is the BOTTOM pin (the one under the the rocker thing, that is hard to get to) as GND soldering is harder.

StanMOI, 2018-04-16:
Dear Kite,
After many tests. I decided to desolder and to solder again the another one "rocker switch".
Then i tested all soldered pins (all OK).
After i updated arduino code and tested with different lines (uncomment only one line, two, three).
For now i find one option that makes the "rocker switch" work's !
It's work's only if i uncomment **all 3 lines**:

```
#define USE_OLD_BUTTON_COMBOS //Changes UP/DOWN to do brightness instead
#define USE_VOLUME_DIGITAL // Use CC1/CC2 or the rocker switch for volume
#define USE_ALT_PINS_VOLUME_DIGITAL // Alternative digital volume pins
```

For my surprise, it's work's not so gently. 
For example, when i push one's UP, i have to wait about 1 second to hear sound changes.
And if i keeping holding to UP or DOWN, after one second sound jump to 100% or 0%.
"joystick calibration" was done just before, but you never can see if it was done well or no cause no any message. It's should be great to see some little message at the end, like "joystick calibration is finish".

Best regards

kiteretro, 2018-04-17:
Ok good, yep all of those should be enabled for it to work. You don't need the first one though, just makes up/down do the brightness instead of volume.

I have a solution to all that, it's not finished but it will be coming soon, and will show on screen those kind of messages!

StanMOI, 2018-04-18:
Thanks Kite!

One day it's work fine, another day not at all (have to reboot).
I tried once to see if it's works in "KODI" and it's doesn't work.
But i'am already in good way.

Can't wait for updates ;)

cobramike, 2018-04-21:
Ok stil not working. Capton tape on it.

Here is the log file.

`Jul  5 11:45:51 retropie rsyslogd: [origin software="rsyslogd" swVersion="8.4.2" x-pid="408" x-info="http://www.rsyslog.com"] start
Jul  5 11:45:51 retropie kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jul  5 11:45:51 retropie kernel: [    0.000000] Linux version 4.14.14-v7+ (pi@retropie) (gcc version 4.9.2 (Raspbian 4.9.2-10)) #1 SMP Sat Jan 20 09:41:50 UTC 2018
Jul  5 11:45:51 retropie kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jul  5 11:45:51 retropie kernel: [    0.000000] CPU: div instructions available: patching division code
Jul  5 11:45:51 retropie kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jul  5 11:45:51 retropie kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Compute Module 3 Rev 1.0
Jul  5 11:45:51 retropie kernel: [    0.000000] Memory policy: Data cache writealloc
Jul  5 11:45:51 retropie kernel: [    0.000000] cma: Reserved 8 MiB at 0x2e400000
Jul  5 11:45:51 retropie kernel: [    0.000000] random: fast init done
Jul  5 11:45:51 retropie kernel: [    0.000000] percpu: Embedded 17 pages/cpu @aef96000 s38720 r8192 d22720 u69632
Jul  5 11:45:51 retropie kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 190820
Jul  5 11:45:51 retropie kernel: [    0.000000] Kernel command line: bcm2708_fb.fbwidth=320 bcm2708_fb.fbheight=240 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:DA:5A:EA vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=5728b712-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait loglevel=3 consoleblank=0 plymouth.enable=0
Jul  5 11:45:51 retropie kernel: [    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
Jul  5 11:45:51 retropie kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Jul  5 11:45:51 retropie kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Jul  5 11:45:51 retropie kernel: [    0.000000] Memory: 741328K/770048K available (7168K kernel code, 548K rwdata, 2028K rodata, 1024K init, 706K bss, 20528K reserved, 8192K cma-reserved)
Jul  5 11:45:51 retropie kernel: [    0.000000] Virtual kernel memory layout:
Jul  5 11:45:51 retropie kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Jul  5 11:45:51 retropie kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Jul  5 11:45:51 retropie kernel: [    0.000000]     vmalloc : 0xaf800000 - 0xff800000   (1280 MB)
Jul  5 11:45:51 retropie kernel: [    0.000000]     lowmem  : 0x80000000 - 0xaf000000   ( 752 MB)
Jul  5 11:45:51 retropie kernel: [    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
Jul  5 11:45:51 retropie kernel: [    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
Jul  5 11:45:51 retropie kernel: [    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
Jul  5 11:45:51 retropie kernel: [    0.000000]       .data : 0x80c00000 - 0x80c89244   ( 549 kB)
Jul  5 11:45:51 retropie kernel: [    0.000000]        .bss : 0x80c90ee0 - 0x80d41734   ( 707 kB)
Jul  5 11:45:51 retropie kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jul  5 11:45:51 retropie kernel: [    0.000000] ftrace: allocating 24618 entries in 73 pages
Jul  5 11:45:51 retropie kernel: [    0.000000] Hierarchical RCU implementation.
Jul  5 11:45:51 retropie kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jul  5 11:45:51 retropie kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Jul  5 11:45:51 retropie kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Jul  5 11:45:51 retropie kernel: [    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Jul  5 11:45:51 retropie kernel: [    0.000021] Switching to timer-based delay loop, resolution 52ns
Jul  5 11:45:51 retropie kernel: [    0.000300] Console: colour dummy device 80x30
Jul  5 11:45:51 retropie kernel: [    0.000323] console [tty1] enabled
Jul  5 11:45:51 retropie kernel: [    0.000351] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Jul  5 11:45:51 retropie kernel: [    0.000367] pid_max: default: 32768 minimum: 301
Jul  5 11:45:51 retropie kernel: [    0.000715] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
Jul  5 11:45:51 retropie kernel: [    0.000736] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
Jul  5 11:45:51 retropie kernel: [    0.001869] Disabling memory control group subsystem
Jul  5 11:45:51 retropie kernel: [    0.001960] CPU: Testing write buffer coherency: ok
Jul  5 11:45:51 retropie kernel: [    0.002470] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Jul  5 11:45:51 retropie kernel: [    0.002902] Setting up static identity map for 0x100000 - 0x10003c
Jul  5 11:45:51 retropie kernel: [    0.003042] Hierarchical SRCU implementation.
Jul  5 11:45:51 retropie kernel: [    0.003737] smp: Bringing up secondary CPUs ...
Jul  5 11:45:51 retropie kernel: [    0.004413] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
Jul  5 11:45:51 retropie kernel: [    0.005113] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[... cut ...]
Jul  5 11:48:30 retropie kernel: [  175.562469] hid-generic 0003:1B1C:1B20.0002: input,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair STRAFE RGB Gaming Keyboard] on usb-3f980000.usb-1.4/input0
Jul  5 11:48:30 retropie kernel: [  175.568955] input: Corsair Corsair STRAFE RGB Gaming Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:1B1C:1B20.0003/input/input2
Jul  5 11:48:30 retropie kernel: [  175.632367] hid-generic 0003:1B1C:1B20.0003: input,hiddev96,hidraw2: USB HID v1.11 Keyboard [Corsair Corsair STRAFE RGB Gaming Keyboard] on usb-3f980000.usb-1.4/input1
`