kiteretro/Circuit-Sword #42: [Enhancement] Ability to turn on/off display

TheOnlyJonto, 2018-04-24:
I would like to be able to turn the display off and on easily in order to save battery without shutting the system down.

jamesalmeida, 2018-05-02:
Maybe turning the brightness all the way down and then going down 1 more can make the screen turn off? Not sure if that's possible but would be a cool way to incorporate the feature.

TheOnlyJonto, 2018-05-04:
Problem is adjusting the brightness from max to min takes forever. I'd rather see this implemented as its own command so I can turn it on and off as quickly as possible.

jamesalmeida, 2018-05-04:
Then maybe holding mode+Left for 3 seconds can skip lowering brightness and just turn off screen. 

Or maybe Mode + Right then Left 

Any other ideas or suggestions on which button combo could be used that’s available?

TheOnlyJonto, 2018-05-04:
Well, as far as I know, left and right aren't used for any mode button combos so it could just be a simple mode+left/right to turn the display on and off. No need to hold it down.

jamesalmeida, 2018-05-04:
I believe by default `MODE+LEFT` lowers the screen brightness and `MODE+RIGHT` increases the screen brightness.

TheOnlyJonto, 2018-05-04:
No up and down control screen brightness.

jamesalmeida, 2018-05-05:
Ah then Left and Right controls Volume.

jamesalmeida, 2018-05-06:
Maybe because my board has HDMI instead of the volume wheel it needs the MODE control otherwise I can't set my volume. The original board had a volume wheel but the newer Circuit Sword has 2 different back boards. 1 with HDMI and the other has a volume wheel instead of the HDMI out. 

He mentions the 2 backboards on the first 2 questions on the wiki page: https://github.com/kiteretro/Circuit-Sword/wiki/FAQ

And you can see the HDMI backboard in the picture in the first post here:
https://sudomod.com/forum/viewtopic.php?f=38&t=5212

TheOnlyJonto, 2018-05-06:
Ah. I do not have the circuit sword. I guess the rear buttons could be used for this function then.

jamesalmeida, 2018-05-06:
Good point. I don’t think the back buttons current do anything with the MODE button.