kiteretro/Circuit-Sword #43: CSO freezes on large wifi transfers without user input

c-hri-s, 2018-05-02:
Discussed here; https://www.sudomod.com/forum/viewtopic.php?f=51&t=5558

To replicate:

1. Initiate a large transfer over Wifi (I tested with a 650MB file)
2. At some stage the CSO will dim the screen
3. At a later stage the CSO will become unresponsive, although you can interact with it via SSH. The power button does not initiate the shut down.

The issue seems to be related to fetching CPU temperature which starts to hang (/opt/vc/bin/vcgencmd measure_temp doesn't return). This also has the side effect of preventing you SSHing into the device as that command is called in the bash profile. If you interrupt the login you are able to successfully ssh into the device to debug, although the device seems to behave somewhat strangely (hanging on CPU temp and other operations). When in this frozen state you are able to successfully transfer data to/from the device via SCP/Samba, and are able to execute other commands.

The issue is not related to Emulation Station, this occurs even if you quit ES and have the device sitting at the console screen.

The issue doesn't seem related to general idling, as if you disable wifi and leave the device untouched it doesn't freeze even after extended idle.

This isn't a critical issue as large data transfers shouldn't be frequent. The most annoying symptom is that you would need to force shutdown the device if you aren't able to SSH into the unit from a PC. This could cause SD corruption.

c-hri-s, 2018-06-11:
Just to confirm this still happens on the retropie-4.4-rpi2_rpi3-wifi_CSO_CM3_20180603-154410 build.

cturczynskyj, 2018-06-30:
I have experienced this same issue myself. I definitely saw my CSO hang when running a script to retrieve temperatures. I was also transferring files and wondered what affect that had on temperatures. Breaking out of the script with control+c worked and running the same script would hang again. This would continue until reboot.