kiteretro/Circuit-Sword #49: New power switch is too close to case for OEM switch

JamesHagerman, 2018-06-14:
I'm posting here because I can't figure out where to post on sudomod.

I finally went to assemble my Circuit Sword and am running into a problem with how the power switch is expected to work.

I'm using HoolyHoo's glue-less bracket: [https://sudomod.com/forum/viewtopic.php?f=58&t=5321](https://sudomod.com/forum/viewtopic.php?f=58&t=5321)

When I install everything, there's almost zero space between the switch lever and the back of the case:

[image]

However, the switch expects a good amount of space between the switch lever and the back of the case so that the lever can enter close to the center of the switch body:

[image]

I had a hard time holding all this together to get a photo, so the issue is a little exaggerated in this photo, but it shows the amount of distance expected between the switch lever itself and the switch body's little ridge that slides on the case:

[image]

So, as far as I can tell, there's basically no way to get this to work unless you:

1. Raise the PCB up with some sort of spacers (but this will push it up against the USB-C port so that won't work)
2. Modify the OEM switch body so that the switch lever can enter closer to the case (but this means the switch lever will probably be exposed out the back of the switch and look pretty ugly)
3. Modify the switch lever itself in some way so its not as close to the body (which will compromise the already-pretty-thin switch lever)
4. Replace the entire electrical switch with one that moves the switch lever closer to the PCB.
5. Just leave the OEM switch body out entirely and reach in with my thumbnail to toggle the power switch

I looked around and couldn't find anyone else having this issue. I've tried this with a cheap cloned case from ebay and an original game boy case and they both have the same problem

Any ideas what might be going on?

kiteretro, 2018-06-14:
Hi, cut the switch plastic shorter to fit properly. Resting the black plastic on some wood and using a sharp cutting knife works well!

closed 2018-06-14 by kiteretro

jamesalmeida, 2018-06-14:
Yeah I chopped the switch shorter on mine and it fits like a glove now.