kiteretro/Circuit-Sword #50: TypeError: ord() expected a character, but string of length 0 found

wardhus, 2018-06-17:
Following the update instructions to enable the volume wheel I get the following error running 

pi@retropie:~/Circuit-Sword $ python cs-configure.py
INFO:root:Program Started
WARNING:root:WARNING: Bad return for command [u]
Traceback (most recent call last):
  File "cs-configure.py", line 395, in <module>
    process_menu()
  File "cs-configure.py", line 307, in process_menu
    'joystick_status': get_joystick_adc(),
  File "cs-configure.py", line 244, in get_joystick_adc
    'joy1_x': convert_data({}, 'u', is_binary=False, length=2),
  File "cs-configure.py", line 192, in convert_data
    return convert_to_decimal(data)
  File "cs-configure.py", line 164, in convert_to_decimal
    for bit in convert_to_fake_byte(byte):
  File "cs-configure.py", line 157, in convert_to_fake_byte
    return bin(convert_to_byte(data))[2:].rjust(8, '0')[::-1]
  File "cs-configure.py", line 159, in convert_to_byte
    return ord(data)
TypeError: ord() expected a character, but string of length 0 found

kiteretro, 2018-06-19:
Hi any update on this?

wardhus, 2018-06-19:
Sorry for the late reply. You assumed correctly. Flashing the Ardiuino fixed the issue. https://github.com/kiteretro/Circuit-Sword/wiki/Configuring-the-software-without-flashing-the-arduino might need some more information because the title said without flashing :)

Anyway, up to the next issue: I installed the backboard with volume control and enabled analog volume in the configuration:

VOLUME INFORMATION
Amp enabled:            1
Current volume:         100%
Analog volume enabled:  1
Analog volume adc:      0

But the volume wheel doesn't seem to work. Any advice?

kiteretro, 2018-06-19:
Great thanks, I'll work on making that clearer (the reference to flashing was that previously to adjust such things you would have to open the code in Arduino IDE and compile and upload, but now you can just use the cs-configure scipt).

Hm, the "Analog volume adc" is showing 0, when you move the wheel and press ENTER, does that value update at all? Can you double check how it is connected to the board, and if you think it's all ok show some pictures of the set up (where bother ends are connected to)?

wardhus, 2018-06-19:
Moving the volume wheel while pressing enter to refresh the value i constantly see a value between 0 and 5. But when i don't move the volume wheel and i refresh it's also a value between 0 and 5. 

Once the value jumped up to 243. I'm guessing the 0 - 5 changes is jitter? and the 243 was a valid value. It seems my keys on the back are also unresponsive so a bad connection could be the issue.

 I'm unable to open up my gameboy at this time. I'll double check the connections when i can. We'll close this issue and I'll open a new issue when i double checked the connections?