kiteretro/Circuit-Sword #51: [Issue] No sound on boot up

janitress, 2018-06-19:
The latest Circuit-Sword image (**retropie-4.4-rpi2_rpi3-wifi_CSO_CM3_20180603-154410.zip**) has the sound disabled by default on boot up (splashscreen).

To fix this this simply edit this script
_/opt/retropie/supplementary/splashscreen/asplashscreen.sh_
and change the line:

`omxplayer -o both -b --layer 10000 "$line"`

to
`omxplayer -o alsa -b --layer 10000 "$line"`

(the change is 'both' to 'alsa')

kiteretro, 2018-06-19:
I made a push this morning to fix this :) looks like they moved the asplashscreen file in 4.4. Here's my commit: https://github.com/kiteretro/Circuit-Sword/commit/d859e67a172e9f0668eed781f9e87b877eb2c812 and you can apply it by running the `sudo ./update.sh`

janitress, 2018-06-23:
great! 
seems to work now.