kiteretro/Circuit-Sword #55: Unable to connect to any WiFi

WhistlerNL, 2018-07-03:
Hi,

Wondering if anyone can point me in the right direction. I'm using Kite's Circuit Sword with latest SD Image.

I'm unable to connect to any WiFi network. I do see all available networks, so antenna works. When trying to connect it ask for the password but entering the PW via keyboard or even using the wifikeyfile.txt don't work. Whatever I do it aways says unable to connect.

Thanks
Kevin

WhistlerNL, 2018-07-03:
I have tried the above but still doesn't work. It delayed boot time with few minutes even and when trying to connect to a network it also takes a couple a minutes before displaying 'connecting' message, When restarting the system the boot up messages reports an fail on interfaces, which wasn't there before so it had to do with the lines I added above.

kiteretro, 2018-07-04:
To confirm, you have tried setting up wifi using the following instructions: https://github.com/kiteretro/Circuit-Sword/wiki/Troubleshooting-(LCD-Power-USB-Software)#connect-using-the-wpa_supplicantconf-file ?

Could you also follow these steps to collect the log file, this will show if there are hardware errors: https://github.com/kiteretro/Circuit-Sword/wiki/Troubleshooting-(LCD-Power-USB-Software)#collecting-the-log-file

WhistlerNL, 2018-07-04:
I will try the wpa_supplicant.conf. But I do have a question about that because in de troubleshooting manual it doesn't say where to put the file. Does it just go into the /boot partition?

kiteretro, 2018-07-04:
Yes, in `/boot` (which is what you see when you put SD into PC). I haven’t done it myself but another user posted it on forum as a solution. It’ll either just work, or you’ll have to go to the WiFi menu and there’s an option like ‘connect with wpa_supplicant file’ or something?

WhistlerNL, 2018-07-04:
Sorry, my pc mounts /boot and /repropie, so thats why I wasn't sure.
The Wifi menu has an import function for wifikeyfile.txt not wpa_supplicant as far as I know.
I did just copy messages over after trying to manually connect to a ssid and the end of the message log reads:

Mar 13 23:07:24 retropie kernel: [   76.657413] RTL8723BS: nolinked power save leave
Mar 13 23:07:26 retropie kernel: [   78.855543] RTL8723BS: nolinked power save enter
Mar 13 23:07:31 retropie kernel: [   83.867864] RTL8723BS: nolinked power save leave
Mar 13 23:07:33 retropie kernel: [   86.065424] RTL8723BS: nolinked power save enter
Mar 13 23:07:38 retropie kernel: [   91.077841] RTL8723BS: nolinked power save leave
Mar 13 23:07:40 retropie kernel: [   93.059218] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
Mar 13 23:07:41 retropie kernel: [   93.351835] RTL8723BS: start auth
Mar 13 23:07:41 retropie kernel: [   93.353786] RTL8723BS: auth success, start assoc
Mar 13 23:07:47 retropie kernel: [   99.274590] RTL8723BS: nolinked power save enter
Mar 13 23:07:52 retropie kernel: [  105.147765] RTL8723BS: nolinked power save leave
Mar 13 23:07:54 retropie kernel: [  106.575617] RTL8723BS: nolinked power save enter
Mar 13 23:07:59 retropie kernel: [  111.587712] RTL8723BS: nolinked power save leave
Mar 13 23:08:01 retropie kernel: [  113.785277] RTL8723BS: nolinked power save enter
Mar 13 23:08:06 retropie kernel: [  118.797751] RTL8723BS: nolinked power save leave
Mar 13 23:08:08 retropie kernel: [  120.995378] RTL8723BS: nolinked power save enter
Mar 13 23:08:13 retropie kernel: [  126.007722] RTL8723BS: nolinked power save leave
Mar 13 23:08:16 retropie kernel: [  128.205475] RTL8723BS: nolinked power save enter
Mar 13 23:08:21 retropie kernel: [  133.217814] RTL8723BS: nolinked power save leave
Mar 13 23:08:23 retropie kernel: [  135.415317] RTL8723BS: nolinked power save enter
Mar 13 23:08:28 retropie kernel: [  140.428118] RTL8723BS: nolinked power save leave
Mar 13 23:08:30 retropie kernel: [  142.625183] RTL8723BS: nolinked power save enter
Mar 13 23:08:35 retropie kernel: [  147.627940] RTL8723BS: nolinked power save leave
Mar 13 23:08:37 retropie kernel: [  149.825487] RTL8723BS: nolinked power save enter
Mar 13 23:08:42 retropie kernel: [  154.837906] RTL8723BS: nolinked power save leave
Mar 13 23:08:44 retropie kernel: [  156.819353] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
Mar 13 23:08:44 retropie kernel: [  157.044244] RTL8723BS: start auth
Mar 13 23:08:44 retropie kernel: [  157.047630] RTL8723BS: auth success, start assoc
Mar 13 23:08:50 retropie kernel: [  163.034686] RTL8723BS: nolinked power save enter
Mar 13 23:08:55 retropie kernel: [  167.307704] RTL8723BS: nolinked power save leave
Mar 13 23:08:57 retropie kernel: [  169.505436] RTL8723BS: nolinked power save enter
Mar 13 23:09:01 retropie kernel: [  173.837907] RTL8723BS: nolinked power save leave
Mar 13 23:09:03 retropie kernel: [  176.045295] RTL8723BS: nolinked power save enter
Mar 13 23:09:06 retropie kernel: [  178.417991] RTL8723BS: nolinked power save leave
Mar 13 23:09:08 retropie kernel: [  180.616822] RTL8723BS: nolinked power save enter
Mar 13 23:09:08 retropie kernel: [  181.055380] RTL8723BS: nolinked power save leave
Mar 13 23:09:11 retropie kernel: [  183.255499] RTL8723BS: nolinked power save enter
Mar 13 23:09:16 retropie kernel: [  188.257822] RTL8723BS: nolinked power save leave
Mar 13 23:09:18 retropie kernel: [  190.455786] RTL8723BS: nolinked power save enter
Mar 13 23:09:23 retropie kernel: [  195.467727] RTL8723BS: nolinked power save leave
Mar 13 23:09:25 retropie kernel: [  197.665125] RTL8723BS: nolinked power save enter
Mar 13 23:09:30 retropie kernel: [  202.677884] RTL8723BS: nolinked power save leave
Mar 13 23:09:32 retropie kernel: [  204.793529] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Mar 13 23:09:33 retropie kernel: [  205.251551] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Mar 13 23:09:33 retropie kernel: [  205.434846] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000808
Mar 13 23:09:34 retropie kernel: [  206.605513] RTL8723BS: start auth
Mar 13 23:09:34 retropie kernel: [  206.607426] RTL8723BS: auth success, start assoc
Mar 13 23:09:36 retropie kernel: [  209.040339] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
Mar 13 23:09:37 retropie kernel: [  209.338127] RTL8723BS: start auth
Mar 13 23:09:37 retropie kernel: [  209.340166] RTL8723BS: auth success, start assoc
Mar 13 23:09:40 retropie kernel: [  212.649962] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
Mar 13 23:09:40 retropie kernel: [  212.922114] RTL8723BS: start auth
Mar 13 23:09:40 retropie kernel: [  212.929883] RTL8723BS: auth success, start assoc
Mar 13 23:09:44 retropie kernel: [  216.540789] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
Mar 13 23:09:44 retropie kernel: [  216.845496] RTL8723BS: start auth
Mar 13 23:09:44 retropie kernel: [  216.847368] RTL8723BS: auth success, start assoc
Mar 13 23:09:50 retropie kernel: [  222.714702] RTL8723BS: nolinked power save enter
Mar 13 23:09:51 retropie kernel: [  223.627385] RTL8723BS: nolinked power save leave
Mar 13 23:09:52 retropie kernel: [  224.850486] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
Mar 13 23:09:53 retropie kernel: [  225.210259] RTL8723BS: start auth
Mar 13 23:09:53 retropie kernel: [  225.218429] RTL8723BS: auth success, start assoc
[... cut ...] pie kernel: [  231.034290] RTL8723BS: nolinked power save enter
Mar 13 23:09:59 retropie kernel: [  231.997914] RTL8723BS: nolinked power save leave
Mar 13 23:10:01 retropie kernel: [  233.980468] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
Mar 13 23:10:02 retropie kernel: [  234.221575] RTL8723BS: start auth
Mar 13 23:10:02 retropie kernel: [  234.223379] RTL8723BS: auth success, start assoc
Mar 13 23:10:07 retropie kernel: [  240.154288] RTL8723BS: nolinked power save enter

kiteretro, 2018-07-05:
To confirm, you are using the latest image here: https://github.com/kiteretro/Circuit-Sword/releases/tag/v1.2.0

And that you can see the wifi icon in the top right corner (altho probably with an "!" on it right?)

Can you disassemble so that you only have it completely out of any case, so that it's the main board + CM3 + LCD + back board attached.. and move it close to your router and try connecting? Note that the keyboard is likely in UK localle, if you password for wifi has funky characters you may want to press F4 and type them into the terminal to see which key is what!

Also, do you have another AP that you can connect to? For example setting your phone as a hotspot?

WhistlerNL, 2018-07-05:
Yes, I am use the latest image and I do have the wifi icon with an "!" on it. It displays my ssid's and the surrounding neighbours ssid. I am about 8 feet removed from the AP.

I will try again right next to the AP and as soon as I have some time take the board out and try it again.

I am still in the mids of my second build so will also check if I have the same issue with another board.

Thanks for a the input and suggestions so far! Will keep you posted