kiteretro/Circuit-Sword #56: Wifi Issue after update

Reptarr117, 2018-07-07:
Anybody losing Wifi function after Retropie Update? followed the how to update via the wiki but now Retropie no longer recognizes the Wifi Chip, gives me the no wlan0 interface detected error. I've found that after formating and redoing a stock image fixes it. 

I assume that the Driver is no longer present but cant figure out how to re add it if i dont have wifi.

kiteretro, 2018-07-07:
Which update were you doing? If you are upgrading the kernel you will loose WiFi because it is a kernel module that I had to specially compile.. so as long as you don’t upgrade the kernel this won’t happen

Reptarr117, 2018-07-07:
oh okay so do i need to re-format with a fresh image or can I still save it.

and it was just the retropie setup (update retorpie) function in the ES menu.

kiteretro, 2018-07-08:
Right, i guess that must do a kernel update.. I'm looking in to a way to be able to install the wifi drivers even after a kernel update.. the trouble is it needs wifi to get the latest packages! So we'll see.. if new major versions come out I do make a new image for them, or if there were an update that was a real game changer then I'd do a special one.. so if there's something specific that you think it needs then let me know.

It would be possible to script the kernel module install, and would require either to put SD into a Pi3, or use a USB wifi adapter. When I have some time I could do that.

umbynos, 2018-09-22:
If you want to upgrade but not the kernel you can use the comand `sudo apt-mark hold raspberrypi-bootloader raspberrypi-kernel  raspberrypi-kernel-headers`. Basically it should not upgrade the package passed as argument. You can check the status of the pakages with `dpkg --get-selections | grep hold`.
Doing `sudo apt update && sudo apt upgrade -d` should return something similar to: 
```
[...]
The following packages have been kept back:
  libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 raspberrypi-bootloader
  raspberrypi-kernel raspberrypi-kernel-headers
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
```
(`-d` flag is "download mode", to install the updates you have also to do `sudo apt upgrade`)

VictorioBerra, 2019-05-10:
@Reptarr117 Did formatting fresh fix your issue? The same thing happened to me but fresh format did not fix it.

Reptarr117, 2019-05-10:
I't did. I wiped the micro SD card, in partitioned, and formatted the card.
Re installed the OS that Kite had on his github. Fixed it. Make sure you go
into the disk management to do that. Worked fine after that.

VictorioBerra, 2019-05-10:
Go into disk management and format you mean?

Reptarr117, 2019-05-10:
Yes might have to clear the partition too. There are tutorials online.

njco90, 2020-04-12:
where can i find a tutorial ? my WiFi has gone out as well. I've tried to google search but haven't gotten any results. or i just don't know the terminology to search correctly. could you kindly post a link?

kiteretro, 2020-04-13:
@njco90 did you do an update? If you have another SD and you re-image with the versions in this git repo, does wifi work again?

holysoles, 2020-04-25:
I just assembled my kit and on a fresh install (rpi-4.4-rpi2_rpi3-wifi_CSO_CM3_20191118-200043) ran a full update (including kernel) not knowing any better, and got the wifi error. I then reformatted with gparted and flashed with etcher, didn't fix it. I zeroed the sd card with dd and flashed with etcher which worked, until i rebooted the pi (no updates installed) and still no wifi. Any suggestions?

holysoles, 2020-04-27:
update, I have wifi on some reboots, on others no wifi. Works for now I guess, since I primarily wanted it for setup.

kiteretro, 2020-04-30:
when wifi stops working on a boot, if you do the combo to DISABLE wifi, wait a FULL 10 secs, then button combo to ENABLE wifi.. does that fix it or is it always a reboot to fix?

Depending how you use it, wifi is only used for config and rest of the time combo-ing it off saves battery