kiteretro/Circuit-Sword #64: First boot only black screen, no boot?

cupowater35, 2018-10-03:
So I finally got the time to start building, and in testing the electronics the first time I have run into a problem where the screen is only black...  I read through the other issue (#57) But I noticed that the PGOOD light does not come on.  What can I do about it? or What am I doing wrong?  Any advice will help.
[image]
[image]
[image]
[image]
[image]

kiteretro, 2018-10-03:
Hi, what image did you burn to the SD card? From the looks of it.. the USB HUB LED is on, meaning that the board has booted into some kind of kernel. However the lack of PGOOD might mean it isn't running the custom image that can be found on the 'releases' tab, could you confirm that?

cupowater35, 2018-10-03:
I was using a card that was built for a pi 3 and already working, I will try the version on the releases tab.  Is it any more than flashing the version?

kiteretro, 2018-10-03:
Nope that’ll be it, in order for the lcd to work it needs some special settings which are already on the image in the releases tab :)

cupowater35, 2018-10-03:
Thank you, your work is amazing and I look forward to future builds.

cupowater35, 2018-10-03:
Any idea on why the screen flashes every few seconds?

kiteretro, 2018-10-03:
Are you able to create a short video showing this? Also does the same happen if you use the 2nd most recent image (rather than the very latest pre built image)?

cupowater35, 2018-10-03:
Of course when I go to take a video it works normally, If I do get a change to capture it I will post it.  Basically the screen would flash black ever 3 or 4 seconds.

cupowater35, 2018-10-03:
[image]

kiteretro, 2018-10-03:
Wow, weird! Does it only happen on that game? Does it happen if you exit to the retropie menu? What happens when you adjust the brightness down a bit?

cupowater35, 2018-10-03:
seems to work fine on any brightness under 40%  same symptoms on all games and menus

kiteretro, 2018-10-03:
Have you soldered the 'backlight boost' solder jumper? It's underneath where the CM3 module is.

If you remove the CM3 module (after having put the brightness back up to full) and power it on, the screen should be black/grey .. does it show the same here?

If you can get it to do this in this state, and you have a multimeter, could you measure the voltage at this point (to gnd) and see if when the flicker happens that the voltage drops, and also what the voltage is! Your meter may not be quick enough to see the dip, but the voltage would help:

[image]

Also while you're there... do those chips get really hot (near that red circle)? Be careful putting your finger on it because there is a chance that it's super hot!

My suspicions so far are a problem with the backlight chip, either getting too hot and overcurrenting, or just being faulty!

cupowater35, 2018-10-04:
I will let you know as soon as I get a chance to test it out.

shorage, 2018-10-06:
> 
> 
> I was using a card that was built for a pi 3 and already working, I will try the version on the releases tab. Is it any more than flashing the version?
> 
> Worked! Sorry to have missed the big step of the version you have made. Thank you again for your quick response, as well as all your good work.

Which image did you use? I tried the latest image and it is exhibiting the same behavior your described.

Thanks

Frank

kiteretro, 2018-10-06:
> Which image did you use? I tried the latest image and it is exhibiting the same behavior your described.
>
> Thanks
>
> Frank

Hi Frank, could you please create a new issue and include pictures of it on in the state (similar pictures to the ones on this issue). Could you also detail how you write the image to your SD? I'm going to assume you followed every single step on this page: https://github.com/kiteretro/Circuit-Sword/wiki/Flashing-Software-onto-the-Compute-Module