kiteretro/Circuit-Sword #65: cs-hud fails to run on Retropie 4.3 due to missing libpng16

jarettmillard, 2018-10-03:
I ran the update script to get the new HUD and it fails to run. I am using the Retorpie 4.3 image. The `libpng16-16` package is not available via apt.

```
pi@retropie:~/Circuit-Sword/cs-hud $ ./cs-hud
./cs-hud: error while loading shared libraries: libpng16.so.16: cannot open shared object file: No such file or directory
```
```
pi@retropie:~/Circuit-Sword/cs-hud $ ldd cs-hud
	linux-vdso.so.1 (0x7eefe000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76ed3000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76e9e000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76e23000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76e10000)
	libwiringPi.so => /usr/lib/libwiringPi.so (0x76df2000)
	libpng16.so.16 => not found
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76dca000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76c89000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76c73000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76c59000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76c42000)
	/lib/ld-linux-armhf.so.3 (0x76ee8000)
	libcrypt.so.1 => /lib/arm-linux-gnueabihf/libcrypt.so.1 (0x76c03000)
```

I installed it manually via the instructions [here](https://connortumbleson.com/2018/06/24/libpng16-a-stressful-day/) and the HUD works afterwards.

kiteretro, 2018-10-03:
Hi, that's good information! My apologies I have only tested it on 4.4 version because i'm personally only supporting 'stretch'. I have updated the 'known issues' with information and a link to here: https://github.com/kiteretro/Circuit-Sword/wiki/Known-Issues#issue-11---cs-hud-fails-to-run-on-retropie-43-due-to-missing-libpng16

jarettmillard, 2018-10-03:
Cool, thanks. There's a similar issue with `avrdude` requiring `libreadline.so.7` that I just noticed too, but that's probably a problem with the apt dependencies. I should really just update to 4.4, but I don't want to lose all the configuration I've already done.

kiteretro, 2018-10-03:
Haha yeh totally understand! My method to update isn't perhaps the best and needs improving for sure, the way in which i build the images means i can't use apt.. and so i `dpkg -x <package>` which obviously doesn't do dependencies (i've resolved them myself) and also doesn't care for the version or release that you are running.. If I were you i would definitely make a backup of your entire SD, it isn't too hard to copy all the settings across as long as you have a backup, but save it for some down time as it might take a while ;) if there's anything you think that i can do to help then let me know

jarettmillard, 2018-10-03:
That's what I'm planning to do. It would be neat if the base image was on a read-only partition and everything else got smushed on top with something like OverlayFS. It would probably be a lot of work to set up on your side, but might be a pretty clean solution to the billion messy folders all over the filesystem RetroPie uses for configuration.