kiteretro/Circuit-Sword #66: Grey colours are purple in LCD (gamma issues?)

piklz, 2018-10-05:
Happens with the 320 and 640 lcd screen

kiteretro, 2018-10-05:
Could you post a picture on the ‘tester’ screen that shows the colour bars?

piklz, 2018-10-05:
[image]

kiteretro, 2018-10-05:
> i originally purchased the 1.2 you send me a lower version replacement it says 1.1e but i assumed it wouldnt have the lcd blob solder issues?(remeber i destroyed the first boards lcd connector trying to fix multple blobs!)

Ah jeez looks like I dropped the ball on this one.. could you send me an email with your address and I’ll send a replacement ASAP! No need to send that one back. This one will not have this issue! Really sorry about that..

piklz, 2018-10-05:
hey Kite, I replied to our conversation via kite@kitesitemshop.com with my address many thanks again for helping out so quickly.
p.s  -none the less, i've enjoyed playing the games on the gameboy even in this state and the new hud is sublime ! keep up the good work- @pixelpiklz

kiteretro, 2018-10-05:
Fantastic :) all received and ready to be sent on Monday! That's good to hear, should just be a straight board swap when it arrives to you ;) I'll put the updated firmware on the board which actually improves the colours on 640x480 (just committed that this morning) so you should be pretty good to go!

closed 2018-10-05 by kiteretro

piklz, 2018-10-23:
new cm3 fixed this colour issues!thanks kite colours are great

kiteretro, 2018-10-23:
Fantastic :) that's good to hear, and especially so seeing as it wasn't my board bu the CM3 itself! :tada: although that wasn't expected..