kiteretro/Circuit-Sword #67: modifying battery/wifi/sound icon location

Olaskan, 2018-10-05:
Method in wiki `sudo nano /home/pi/Circuit-Sword/cs-osd/cs-osd/config.ini` doesn't work. It appears cs-osd has been removed. I'm trying to modify three things relating to what I assume was cs-osd:

1. Move the icons (battery/wifi/sound)
2. Stop the icons from appearing until the splash screen is finished
3. Stop the K logo (kiteretro logo) from appearing during splashscreen (or all togther)

Not sure what to do, any advice would be really appreciated. Thank you.

kiteretro, 2018-10-05:
Hi, the new cs-hud (which replaces the cs-osd) currently doesn't have the same file to change the variables. I've tagged this as 'enhancement' but really it's a regression in functionality.. the code already has the ability to change variables however doesn't have the method to be able to set it from a file. I have planned for this work to happen in the future so moving the icons will come soon!

There are some additional features that will allow hiding icons and such.. for the meantime if you cannot live with this then you can downgrade to an older commit. Is the icon moving specifically for the 640x480 LCD? If so I can make some temporary fixes.

For the other issues, I can look to put a delay timer in the service startup so that everything happens a little later, assuming this is to avoid stuff showing over your startup video?

Olaskan, 2018-10-06:
Hi Kite, yeah it's mainly to stop everything from showing up during the splash screen so it's a bit more seamless. Since adjusting overscan the icons are in a better position but an adjustable delay timer for the osd and kiteretro logo would be really good. 
I'm just using the 320x240 LCD

timemaster5, 2019-04-14:
+1 I need it to adjust icons for 640 LCD :)

idoregesz, 2020-01-27:
+1 need adjust the icons for the new 640x480 LCD! Take a look to the attached pics: 

http://www.kepfeltoltes.eu/images/2020/01/27/40920200125_001725.jpg
http://www.kepfeltoltes.eu/images/2020/01/27/12520200125_001736.jpg

Need to down lower few rows the battery and the wifi icons!

idoregesz, 2021-06-21:
hello again! is there any update with the battery icon moving issue? any plan about when available the next milestone? thanks!

oki-dingo, 2021-07-20:
Kite, i just received my kit with the HD screen. Unfortunately the wifi and battery icons are in the top right corner under my lens. I would also like to move it down a bit.

oki-dingo, 2021-07-20:
Kite, i just received my kit with the HD screen. Unfortunately the wifi and battery icons are in the top right corner under my lens. I would also like to move it down a bit.
[image]