kiteretro/Circuit-Sword #68: No boot / LCD Powers up, nothing on screen

shorage, 2018-10-06:
So i am doing some testing and i am not getting an image on the LCD screen. It appears to power up but i am not getting a signal.

I tried these images
retropie-4.4-rpi2_rpi3-wifi_CSO_CM3_20181003-070618.zip
retropie-4.4-rpi2_rpi3-wifi_CSO_CM3_20180603-154410.zip

I used win32 disk imager and wrote both using [these ](https://github.com/kiteretro/Circuit-Sword/wiki/Flashing-Software-onto-the-Compute-Module) instructions.

Do i need to have a battery plugged in for it to boot? That is the only thing i can think of.

Thanks

Frank Costa

[image]
[image]
[image]
[image]
[image]
[image]

kiteretro, 2018-10-06:
Thanks for the images :) so the CM3 isn't booting, noted by the PGOOD led not being on, and also the USB HUB led not on. Power looks good (no battery is fine, you can all the power LEDs on).

Could you try inserting and removing the SD card from the slot about 5 times or so in a row incase there is flux on the pads.

Failing that, could you remove the SD card and look really closely at the SD card slot on the board. You'll see that the slot has some holes in the top that allow for some inspection. You should be able to see the internal pins that are then soldered underneath to the PCB. Ideally if you could get some macro shots of this and attempt to look yourself; we're looking to see if there is actually soldered on all the pins and that it looks melted and attached.. I appreciate that this is quite difficult and definitely needs "an eye for it" but with this kind of issue, if it isn't the SD card itself then we're getting in to difficult to debug places ;)

Let me know how that goes! I assume that when you now put the SD into your PC, it appears as like a 60mb small drive with a number of text files in?

shorage, 2018-10-06:
So it feels like their is this "rough" spot when i insert the sdcard. i inserted multiple times and every now and then, their is a rough spot. I am guessing their is something lifting inside the slot.

I really cant see anything, it is such a small space and i cant seem to find my "helping hands"

I have attached some pictures that i took, its the best i could do. I do have a second CircuitSword you sent me, i will test with this one and see if their are any changes. Looking at the sdcard in windows, it appears everything is in order.

[image]
[image]
[image]
[image]
[image]

shorage, 2018-10-06:
The second circuit sword powered right up and is booting.
I swapped the main PCB with all the other components i was working with.

shorage, 2018-10-06:
[image]

kiteretro, 2018-10-06:
Ok that's awesome :) thanks for that, yeah it's really hard to see, and right now it's only a suspicion. Could you please send me an email and we'll sort out replacing the board and hopefully I'll be able to debug and report back here what the actual problem is! I have another customer with the same kinda thing sending next week too so hopefully with those two i'll be able to offer advice and also prevent it happening in the future!

shorage, 2018-10-06:
frankanthonycosta@gmail.com

Brouwser83, 2018-10-07:
Morning,

Hope it's ok i am replying to this issue. Please let me know if you want me to open an new one. Also i can provide pics if needed.

I also seem to have this exact problem. Received my board yesterday and did a test this morning, but also only a black screen. It flashes real quick sowing the Raspberry's colorful boot screen for a millisecond but then it turns black.  

Tried both with and without battery and also 2 different images. Also tried a different SD cards. 

The leds all light up including the 5 power leds. I did notice the top led is a little less bright then the others (same as in the pictures above) but i am not sure if this has anything to do with it. 

Thanks in advance. 
Sjoerd Brouwer

kiteretro, 2018-10-07:
@Brouwser83 Morning! Could you open a new issue please, your problem is slightly different because you actually get to see the Pi logo (this issue here doesn't get that far). Meaning yours does boot further along so it might be something else! Could you take some pics and also ideally take a short video of it happening? I'd ideally like to see it with all the LEDs and screen visible in the shot, thanks :)

type11error, 2018-10-10:
FYI I had a similar issue, re soldering the pins with a fin pitch iron on the SD card seemed to fix the problem and it's now booting