kiteretro/Circuit-Sword #69: No Boot / Black screen

Brouwser83, 2018-10-07:
Hi,

Received my board yesterday and did a test this morning, but also only a black screen. It flashes real quick sowing the Raspberry's colorful boot screen for a millisecond but then it turns black.
I made a movie, but its almost not visible. 

Tried both with and without battery and also 2 different images, screens and i tried a different SD card.

The leds all light up including the 5 power leds. I did notice the top led is a little less bright then the others but i am not sure if this has anything to do with it.

[Quick youtoube movie](https://youtu.be/wmUWKBD8l8M)

Some images:

[image]
[image]
[image]
[image]
[image]

Please let me know if you need anything else. 
Sjoerd

kiteretro, 2018-10-07:
Thanks for all the info, so your issue is the same as the other one, the screen flash is just the backlight powering it on, not the 'rainbow pi boot' screen thing. Your message previously confused me because of that but it's ok :)

Do you have a multimeter? If so could you measure the voltage while on, on the other side of the board, on the "1V8" test pad?

Assuming that it does read 1.8v, and that when you place your SD into your PC that it appears as a 56MB drive (this shows the image was written properly). Then could you please send me an email and I'll send you a replacement and I'll also include details on returning the faulty one.

kiteretro, 2018-10-07:
As an FYI I have added this as a known issue for now seeing as I've had a total of about 5 people reporting this: https://github.com/kiteretro/Circuit-Sword/wiki/Known-Issues#issue-12---no-boot--black-screen-only

closed 2019-01-06 by kiteretro