kiteretro/Circuit-Sword #71: hdmi out connected crashes retropie/emulation also sound not working still

piklz, 2018-10-13:
using latests premade image this week (retropie-4.4-rpi2_rpi3-wifi_CSO_CM3_20181003-070618.zip)i tried One-time reboot and also when i leave cable connected it crash more often than not the boot process and nothing appears on tv except gibberish colours
( i think behind the scenes emu station sometimes is loading fine as i can mostly shut down using the main  switch and i see the lights go out as normal.. )
just the display portion on both lcd and tv either freezes or crashes with funky colours or when it did work once only i had no sound anyway .

kiteretro, 2018-10-13:
Weird, when you plug in HDMI when using the internal screen normally, are you able to see anything on screen? (you should be able to, albeit at a low res, because there is a cloning service running that enables this). If it doesn't, do you have another TV you can try on?

piklz, 2018-10-15:
[image]
I will retry the ribbon maybe it's loose or dirty in there

kiteretro, 2018-10-15:
Ok so looking at the resolution that is the 'display cloning' .. what happens when you select "one time reboot from HDMI" ? Same thing?

Yeah clean the connection, make sure your HDMI connector is plugged in ALL the way, you may need to glue in some supports to the back board to make sure it is pushed against the case

piklz, 2018-10-15:
Yes happrns with one time as well .Ok I'll do that and let you know the results cheers kite

piklz, 2018-10-23:
mostly successful !Kite this issue was remedied with the new cm3 -thanks !

 (still no sound in either boards or either cm3) so maybe to get sound out on hdmi i need to tweak some raspberry /emustation configs? but wow what a diff ! 
- colours on lcd smooth  and bright  also finally saw one-time-hdmi for first time (without sound though) and that looks so crisp+hd compared to clone mode! i can see myself also playing it this way now!
So any help with sound configs so i can finally say hardware is fine and start updating the sd with with loads of systems!!yay

[image]
[image]

kiteretro, 2018-10-23:
Great :) do you have a normal Pi or Pi3 and have had audio working? Might be good to try that first? Otherwise..

The file in `/boot/config.txt` (or `config.txt` if plugging in to PC) has these lines highlighted which apply to HDMI: https://github.com/kiteretro/Circuit-Sword/blob/master/settings/config.txt#L50-L75

NOTE: That every line that you wanted enabled MUST have a `#` at the start.. and any other line that is a comment or shouldn't be enabled MUST have two `##` at the start! This is very important because when you run 'one time reboot to hdmi' what it will do is remove 1x of the `#`'s from each line within that section.. so any comments must have the double `##` .. you can see the example in that link above, and it shows the same where every line with 1x `#` will be enabled on reboot :)

You MIGHT find changing the line `#hdmi_drive=2` to `#hdmi_drive=1` might help (changing the 2 to a 1) ?

piklz, 2018-10-23:
cheers for that i went iand tried those config changes heres a list of things of the events i noticed when i try different things..

my normal pi3 i have work with hdmi sound out on both monitor and tv (and tested additionally with  usbsoundcard stick type  (cheap chinese one we all buy for the pi for example)

tried volume:100 still no sound on tv/monitor (inclone mode speaker form gameboy always outputs sound):

- - **if hdmi cable connected before switching on CS  = no lcd** frozen/crashed - cant switch off by its own needs mode button nothing on lcd or tv
- - **if hdmi cable inserted after emustation has loaded = hdmi clone out works** and lcd works gameboy speaker sound also works  **(but no sound on tv/monitor)**
- - **if  one-time-hdmi used it cant have cable already inserted as usually doesnt output on either tv** or gameboy lcd (but CS shutdowns fine without mode button killer)
- **if  one-time-hdmi used and i wait till emustation loaded  and then inserting hdmi** cable tv/monitor **image works looks** great in fact **(but no sound on tv or gameboy  speaker now.)**
- this happens with or without changing hdmi=1 or2 in the config at the moment

Kite hope this helps to wittle down what it could be, ill keep testing diff combos..., cheers piklz

kiteretro, 2018-10-23:
Ok, could you try this:

1. No HDMI plugged in
2. Press 'one time reboot to hdmi'
3. As soon as the screen starts to face, turn OFF the power switch! At this point, the CS should be switched OFF and nothing on screen/lights/etc
4. Plug in HDMI
5. Slide power switch ON

The expectation is that it should boot up in HDMI mode now.. if it doesn't or after 2mins nothing has happened, do the reset and can you upload the file `/var/log/messages` and `/var/log/systemd` (i'm pretty sure that's its name, it might be `/var/log/journalctl` though, can't remember!)

piklz, 2018-10-24:
Gave it a go- like you stated it boots up fine into hdmi hires mode (720p i think) and the screen on lcd is now not white (just on)which could be seen as a bonus. I'm gonna boot it this way now on.
   But no sound on tv/monitor still- also tried a new set of mini HDMI cables same thing, also noticed the hud crashed midway while i was loading diff games to test never saw that before ,not sure what happen'd but picture fine just no sound 
(also no headphone sound? but thats prob supposed to be like that) cheers -piklz

kiteretro, 2018-10-25:
Yeah so it sounds like everything is set up right (no headphone sound means sound is not pointing at the USB chip, which is good) .. the only thing I can suggest is try some other TV types.. it might be a bad connection on the FPC cable but seems unlikely, but i guess that's the only thing left? If this is a deal breaker, I can take a look and debug with my known working configurations!

piklz, 2018-10-25:
I think I worked it out going to redo the order of things and report back today..

piklz, 2018-10-25:
Sounds is on tv yay !went into retropie audio settings and forced HDMI, suddenly sound appeard in one time hdmi mode!(I swear I did this with prev as card I had these issues on)  made another sd card with a fresh cs image and tried neogeo again and that outputs sound and images too which my first sd didn't for whatever reason ..when I went back to normal LCD on gameboy ..the sound came.from speaker /headphone so all working good !phew.

 It looks so great with crt shaders and sound is stable .  Soundwisr it's another step to do before you go hdmi route but it's working fine .

piklz, 2018-10-26:
Cool sounds good! any automagic from your cs scripts would be great ! One time HDMI mode  is def an awesome thing to have .')