kiteretro/Circuit-Sword #73: Buttons

TigerClaw33, 2018-10-25:
Hi, 
I have problems with the buttons. If I push START or want to select other consoles wieht RIGHT or LEFT it jumps like I would push the button 2 or 3 times. I changes all rubbers with other one and also with original one but nothing changed :(

TigerClaw33, 2018-10-30:
crazy thing but this was the solution. cleaning with alcohol.normally I clean all PCBs before mounting excepting this time .. thanks

closed 2018-10-30 by TigerClaw33

kiteretro, 2018-10-30:
Nice, I will be sure to reference this to others! I suspect during manufacture it has some residue of flux remaining! In future i'll make them all ENIG finish (gold) which should be better (but likely will still need cleaning first!)