kiteretro/Circuit-Sword #74: LCD powers on but Pi doesn't boot

davidpic76, 2018-10-26:
I have two Circuit Swords that I have just started testing.
One boots up perfectly but the other one doesn't boot but the LCD powers on.
I switched the compute module, screen and SD card from the working one and got the same result so I think its the board that has the issue.

Any ideas?

[image]

davidpic76, 2018-10-26:
Rotated the image!

[image]

davidpic76, 2018-10-26:
This looks like a dupe of https://github.com/kiteretro/Circuit-Sword/issues/69

kiteretro, 2018-10-26:
Indeed, sorry about this, there is a ‘known issue’ already, could you send me an email to kite@kitesitemshop.com and include your full address and I’ll sort a replacement and give you a return address (and I’ll pay for the return shipping back to me)

kiteretro, 2018-10-26:
Oh looks like you already have ;)

davidpic76, 2018-10-26:
Haha no problem

My address is

David Pickford
2045 Jonathan Ave
San Jose, CA, 95125
USA