kiteretro/Circuit-Sword #75: Button board not responsive

Reptarr117, 2018-11-20:
I have put together, now my third, CM3 build. and I'm getting very poor response from my D-pad and my A,B,X,Y buttons. even my shoulder and trigger buttons are not responsive (using Hooly Hoo's 4 button universal back board) I have updated and flashed the Arduino. I have downloaded and updated the circuit-sword. I have tried everything that my limited knowledge of the subject can think to do. I know its not a OS issue (bad lag) because my keyboard controls the menus just fine. I have to press super hard in order to get a button to work. I have checked the hardware ( made sure the contacts were clean) checked on the actual D-pad and buttons and even the silicone pads underneath. everything feels normal (nice responsive clicks in and out nothing sticks and the case is nice and tight so I know that contact is being made). Any help would be appreciated. Unsure if I have just a faulty board or its a software issue. (leaning towards software due to both the sword and hoolyhoo's board are giving the same "need to press hard to do anything" issue. 
Thanks in Advance!
Reptarr

kiteretro, 2018-11-20:
Hi, on the main board there is an LED "ARD STS" which is a yellow one in the middle between the CM3 and the headphone jack. Under normal use it should be dimly lit (it's actually blinking at 60Hz) but when you press the MODE button it will blink brightly at about 5Hz .. Under normal use, it may occasionally 'blip' brighter for a short amount of time, maybe once every 5 secs or so? This is usually caused by waiting on the OS to process the USB events.. 

Could you please describe the behaviour of this LED as best you can?

Could you also grab these log files and I can see if there is anything reporting errors in the software https://github.com/kiteretro/Circuit-Sword/wiki/Troubleshooting-(LCD-Power-USB-Software)#collecting-the-log-file

Reptarr117, 2018-11-20:
the LED appears to be, instead of starting dim and blinking bright, consistently bright while blinking dim every half a second or so. no changes to it at all just the same pattern of bright to dim. back to bright. if that makes sense. so the opposite kind of "blip" you described. 
only a syslog is available after i do what the wiki told me to do. hope this helps. 
Thx again Kite,
Reptarr

[syslog.txt](https://github.com/kiteretro/Circuit-Sword/files/2601600/syslog.txt)

Reptarr117, 2018-11-20:
also to note. pressing the mode button doesn't effect the LED even tho it comes up correctly in the UI on the screen.

Reptarr117, 2018-11-20:
got it to work here is the message.txt
[messages.txt](https://github.com/kiteretro/Circuit-Sword/files/2601657/messages.txt)

kiteretro, 2018-11-21:
Hi, thanks for the info, looking at the logs everything is perfect and no sign of anything (USB) playing up at all.

Q. When you press the mode button? Does the menu pop up as usual in order to change things like volume etc?

If you run the configurator (https://github.com/kiteretro/Circuit-Sword/wiki/Configuring-the-software-without-flashing-the-arduino#running) and see all the output that appears on the screen.. the line `Mode pressed: 0` does it show that? And if you then HOLD the mode button down and then press `ENTER` (which will refresh that output text) does it now show `Mode pressed: 1` ? You might be best enabling SSH and using PuTTY or similar to remotely log in to the Pi.

If that does all work as expected, then it has to be something hardware to do with the arduino chip and I'll replace it for you if you send me an email (kite@kitesitemshop.com). If the 'mode button' doesn't work at all, then there are one or two things we can try to bring it back to life.

Reptarr117, 2018-11-23:
sent you an email