kiteretro/Circuit-Sword #78: What happened to the Wiki?

jamesalmeida, 2019-01-10:
Seems like a lot of the Wiki info is gone. Has it been moved? I'm hoping to update my board to most current software but don't see anything in the wiki about it anymore.

kiteretro, 2019-01-10:
It’s in the process of being updated and that info will be in there hopefully end of tomorrow, but every day something is being added and streamlined :) it was all a bit disjointed and was only patched each time that I edited it.. doing a proper job this time round!

JustinBenedick, 2019-01-12:
please let me know when its all back up.  A friend of mine has some concerns about his cso and we wanted to see how its put together.

kiteretro, 2019-01-14:
The majority of text is complete (and condensed to the two main pages). Images will be going up hopefully this week

kiteretro, 2019-01-18:
The wiki is 98% updated. There are a few images in the setup process that will be updated slightly, other than that the content is there!

closed 2019-01-18 by kiteretro