kiteretro/Circuit-Sword #81: After update, power switch not working and no HUD

djbriane, 2019-02-20:
Just ran the update script on my GBZ (Circuit Sword CM3) and while everything seems to have worked properly, I no longer see any HUD on my screen and flipping the power switch does not shutdown the system (I have to use the mode button to force kill it).

Are there any debugging steps I can take to determine why the safe shutdown / HUD are not working? 

(if it matters, this is not the latest version of the Circuit Sword, it's one I built around a year ago)

kiteretro, 2019-02-20:
Hi, could you please collect and upload the log files: https://github.com/kiteretro/Circuit-Sword/wiki/Circuit-Sword-FAQ#how-do-i-collect-the-log-files (you can use something like pastebin to place the files, or email them to me).

Depending how old it is, you may need to update the microcontroller to support some of the newer stuff: https://github.com/kiteretro/Circuit-Sword/wiki/Circuit-Sword-FAQ#how-do-i-reprogram-the-microcontroller

djbriane, 2019-02-20:
Attempt to update the microcontroller and got this:

```
avrdude: error while loading shared libraries: libreadline.so.7: cannot open shared object file: No such file or directory
ERROR: Failed to flash Arduino [127]
```

kiteretro, 2019-02-20:
Could you try running 
```bash
sudo apt-get update
sudo apt-get install libreadline7
```

kiteretro, 2019-02-20:
Failing that, run:

`sudo apt-get install avrdude`

djbriane, 2019-02-20:
`messages`:
https://gist.github.com/djbriane/b938a422a397c520d7555bad8c842682

`syslog`:
https://gist.github.com/djbriane/9b26df87a17db22d57b04fbcfd54925a

kiteretro, 2019-02-20:
The issue is:
```
cs-hud[403]: /home/pi/Circuit-Sword/cs-hud/cs-hud: error while loading shared libraries: libpng16.so.16: cannot open shared object file: No such file or directory
```

This is because you are using the older base image, the newer one is based on 'stretch' which does a way with loads of things.. 

You can fix it by following this to get libpng16 installed: https://connortumbleson.com/2018/06/24/libpng16-a-stressful-day/

The avrdude on should be solved by the `sudo apt-get install avrdude` command

djbriane, 2019-02-20:
Okay - I followed the steps in that article to install `libpng16` and then I installed `avrdude` via apt-get. However, when flashing the microcontroller I now see the following error:

```
avrdude: error at /etc/avrdude.conf:1115: programmer type jtagice3_updi not found
```

djbriane, 2019-02-20:
Okay - good news. Ran the update script again and the HUD is back and safe shutdown is now working. Still didn't have any luck flashing the microcontroller but I think I'll leave well enough alone until I'm willing to make the jump to stretch.

djbriane, 2019-02-20:
One more note on the avrdude issue - I'm back to seeing the issue where it cannot find `libreadline.so.7`. Also, `libreadline7` is not available via apt-get so probably need to compile that from source?

jamesalmeida, 2019-02-20:
I'm having the same issues. Would it be easier to just reinstall everything on my SD card?

kiteretro, 2019-02-21:
Honestly, it would be easier to do that :) I no longer support those older images and I don't have them around to be able to test and give you guys real answers!

If everything works (including the on screen keyboard) then there isn't a need to flash the microcontroller. There isn't anything else that has been updated on it, the last thing I did was add the microcontroller code to work with the onscreen keyboard.

Otherwise, backup as much as you can and re-flash to the latest :) hint, Paragon EXTFS is a neat tool for mounting linux file systems within Windows.

djbriane, 2019-02-21:
One suggestion would be to have the upgrade script check the current OS version and give a warning / fail before it starts making changes.

jamesalmeida, 2019-02-22:
@kiteretro I've never seen an onscreen keyboard on mine. Do I need to flash something on the board to get that to work or will reinstalling from scratch on my SD card fix that?

kiteretro, 2019-02-22:
> One suggestion would be to have the upgrade script check the current OS version and give a warning / fail before it starts making changes.

Yep would make sure, it was quite some time ago that it changed base versions! This project is very much a 'only upgrade if you know there is something new that you want' and you're right I should do that.

> @kiteretro I've never seen an onscreen keyboard on mine. Do I need to flash something on the board to get that to work or will reinstalling from scratch on my SD card fix that?

Re-installing will do it, when you hold the MODE button and press START it'll enable the keyboard with the new version of software.

jamesalmeida, 2019-02-22:
Sweet. Paragon also has a mac app for mounting extFS drives. Thanks for the tip @kiteretro !

https://www.paragon-software.com/us/home/extfs-mac/

jamesalmeida, 2019-02-23:
@kiteretro So I went ahead and completely reinstalled the software. The HUD is back but now Wifi is not working. I turn ON and OFF with the MODE button but when I turn it on it has an "X" through the icon. 

When I go to Wifi settings and then into "Connect to Wifi" gives me the error `No wlan0 interface detected`.

jamesalmeida, 2019-02-23:
Yeah I fixed the wifi. The tape had moved when I opened my case. So all is good for me now. HUD is back after reinstalling everything and wifi works. Thanks.