kiteretro/Circuit-Sword #83: After update from to 1.3.2 : NO Volume, NO Battery info on screen, Safe shutdown doesn't working anymore

StanMOI, 2019-02-26:
Hello,

After i made update many things not working well.

- Digital Volume note working anymore.

- Impossible to enable the Volume wheel using method from FAQ.
`ERROR:root:ERROR: Bad return for command [u] - Not supported! Update the Arduino code..`
The value stay always to 0
`Analog volume enabled:  0`

- Safe shutdown doesn't working anymore, i have to force it by using : Turn the power switch "OFF" then press the "mode" button.

After some research, i tried to do Arduino update via Terminal:
```
cd Circuit-Sword
sudo apt-get install avrdude
sudo ./update.sh YES
sudo ./flash-arduino.sh
```
When i do `sudo ./flash-arduino.sh`
i got error code:
```
Failed to stop cs-osd.service: Unit cs-osd.service not loaded.
Resetting Arduino..
Reset completed
Flashing Arduino..
avrdude: error while loading shared libraries: libreadline.so.7: cannot open shared object file: No such file or directory
ERROR: Failed to flash Arduino [127]
At this point you may be lucky and in 10 seconds the Arduino will reboot
Othrewise you can try again, or try plugging in to a PC and using the Arduino software to upload
Failing that, you will need to program using an AVRISP
```
Should i download IMAGE and reinstall it again?

Thank

kiteretro, 2019-02-26:
Hi, your issue is exactly the same as issue #81 where you are upgrading from a much older version of the image. The version jump is so large that there are missing libraries and no migration path. Really it is clear that I should prevent the upgrade in this case.. 

Your best bet is to download the latest image and copy stuff across. Please let me know how the digital volume thing works out, there may be some slight code changes that I'll need to give you depending on the hardware version of your board.

StanMOI, 2019-02-26:
Thank you Kite.
I Will do it tonight and give you feed-back 
;)

StanMOI, 2019-02-27:
Hi Kite,
I done all.
After downloader and installed the lasted version of the IMAGE:
update of Arduino worked.
I enabled the volume wheel ON, but it's still not working.
Please can you give me a code to make it works.
Thanks

kiteretro, 2019-02-27:
To clarify, what did you enable? In the configurator there should be an 'enable digital volume', is that what you did? Could you paste the output from the show command here?

StanMOI, 2019-02-27:
yes, and after i check it again.
`Analog volume enabled:  1`
now even mode button not working :(

kiteretro, 2019-02-27:
Ok but you wanted the digital volume rocker? Or have I misread?
Could you post a picture of what it is you are using for volume control :)

Is the battery icon at the top of the screen? (if you've only just run the configure commands, then the battery won't be there until you reboot)

StanMOI, 2019-02-27:
Sorry Kite,
It was my mistake!
I set on Analog volume.
Now i used the right one:
`Digital rocker enabled: 1`
Here is my volume rocker: https://ae01.alicdn.com/kf/HTB16VaoQXXXXXcEXFXXq6xXFXXXN/204932581/HTB16VaoQXXXXXcEXFXXq6xXFXXXN.jpg
After the physical reboot, the battery icon is again on the screen.
Mode button works again, and the digital volume rocker also.

What kind of Analog volume rocker can i use, do you have a link please?

StanMOI, 2019-02-28:
What about Joystick Kite?
Have you any idea why it's doesn't work?
Even after calibration.
Thank's

StanMOI, 2019-02-28:
I found the answer.
It's have to be configure in Emulationstation and Retroarch

closed 2019-03-04 by kiteretro