kiteretro/Circuit-Sword #84: [Dokumentation] Digital Volume Guide for HW V1.1E

hankman13, 2019-02-26:
Hello,

I have the HW version 1.1E and wanted to set up the digital volume. So I can remember that there was an instruction with an picture for cutting and rewiring the digital knob. Can you share this part of "troubleshooting" for the old hardware again? Or is there an archive of the old wiki?

Thank you!

kiteretro, 2019-02-26:
Will do, should have the information up within a week or so and i'll update here

hankman13, 2019-02-26:
perfect, thank you

timemaster5, 2019-04-14:
Is it already done? I am going to be in the same situation. Digital switch arrived last week :)

kiteretro, 2019-04-15:
While not official, this should help: https://github.com/winder/Circuit-Sword/wiki/Digital-Volume-Guide#digital-volume-buttons-rocker-or-buttons

hankman13, 2019-04-17:
Perfect, that is information i'm looking for.
Thank you.