kiteretro/Circuit-Sword #86: D-pad as joystick not persistent

pgzh, 2019-03-11:
When setting "D-pad as joystick" via the mode button, the setting is lost (off again) after rebooting. Volume, brightness, wifi state etc. are saved.
Also, when use D-pad as joystick is ON, the D-pad is recognized as buttons in RetroPie. When it is OFF, the D-pad is recognized as joystick hat.
Considering the name it would make sense to have the D-pad being recognized as joystick hat when the setting is ON and not OFF.
Please have a look into saving and maybe switching this functionality. Thanks!

kiteretro, 2019-03-12:
Hm I see your point, really the option is "Dpad as buttons" rather than anything else! It doesn't remember this because on a fresh boot it may be confusing that navigation doesn't work. The "hat" input is the default and I want to keep it that way because that is the standard direction pad input

pgzh, 2019-03-12:
Since the "hat" input didn't work out for Dosbox I actually changed to button mode and reconfigured the controller in ES, so my navigation actually does not work after rebooting until I switch to button mode...
I haven't found anything not working with this configuration yet... Please reconsider remembering the setting.

kiteretro, 2019-03-12:
If you open the arduino code in the Arduino IDE and comment/remove this line: https://github.com/kiteretro/Circuit-Sword/blob/master/kite-arduino/CS_FIRMWARE/CS_FIRMWARE.ino#L203 it will save the state for you.

To upload the code, you need to set the DIP switches on the CS so that PROG_EN and USB_HUB_PASSTHRU are enabled. Then turn the CS on (it will not boot) and plug in the USB C cable to your PC. If it says unrecognised devices, you will need to plug the CS into a USB HUB and then into your PC (weird, but it works). The Arduino is an "Arduino Leonardo"