kiteretro/Circuit-Sword #87: Anolog volume wheel doesn't work

StanMOI, 2019-04-13:
I definitely have no chance. 
After that I have many problems with digital volume Wheel, I changed to Anologic, 
but my Anologic Volume wheel not working at all.
I can change my volume only with Mode button.
Here is setup:

```
VOLUME INFORMATION
Amp enabled:            1
Current volume:         3%
Digital rocker enabled: 0
Analog volume enabled:  1
Analog volume adc:      221

```
I made Circuit Sword Software updates before.
I reprogrammed the microcontroller with 
`sudo ./flash-arduino.sh`

Sorry Kite,
I definitely have no chance.
I surprising that I'am the only one who have this problem.

kiteretro, 2019-04-13:
If you rotate the vol wheel all the way to what would be 'minimum volume' and press ENTER (to refresh those stats). Does the `Analog volume adc` now read `0` ? If you then rotate it a bit, and press ENTER, does it go up? Can you keep doing this and see if when you get all the way to the top (full volume) that it says `1024` .. or let me know the behavior of this value?

Once you've done that, could you please collect and upload these log files: https://github.com/kiteretro/Circuit-Sword/wiki/Circuit-Sword-FAQ#how-do-i-collect-the-log-files

StanMOI, 2019-04-13:
> If you rotate the vol wheel all the way to what would be 'minimum volume' and press ENTER (to refresh those stats). Does the `Analog volume adc` now read `0` ? If you then rotate it a bit, and press [...]

When I rotate to max (UP), and refresh with ENTER, here is the value:
`Analog volume adc:      237`
It never says `1024`
When I rotate to minimum (DOWN), the value is:
`Analog volume adc:      0`
Max UP, or Max DOWN, the value of **Current volume** , stay allways on :  `3%` 
When I stay on MAX and refresh every 5 second I got every time different values of **Analog volume adc**
When I stay on MIN and refresh, I have the same problem,  values of **Analog volume adc** changes every refresh.

Here is the log file of **message.txt**:
[https://ufile.io/8jd7pxxv]

Here is the log file of **syslog.txt**:
[https://ufile.io/zxb47ccr]

StanMOI, 2019-04-16:
Kite, any solution?

kiteretro, 2019-04-16:
Ok so the arduino isn't reading the analog value properly..

With it all ON and working, can you use a multimeter and measure the voltage between the two highlighted pins on the wheel itself (the bottom one is the GND, and the one above is the output signal):

[image]

If you rotate the wheel while measuring you should be able to get it to go from 0v all the way to 3.3v. If you can confirm that, then the volume wheel is working at least, and that means the issue will likely be in the cable or the connectors at each end.. it's then a matter of probing the middle wire at the connector ends of the board and seeing where it no longer reads the signal.

If it doesn't read between 0-3.3v, instead do the same but with these two points (the bottom one is GND, and the top is meant to be a CONSTANT 3.3v:

[image]

And this is meant to be a constant 3.3v. If it isn't, then work your way down the cable/connectors measuring until it becomes 3.3v.. 

You can see the pads labelled on the connector ends. Basically the pot wheel is a divider of the voltage, and will output down the "AVOL" wire the voltage between 0v - 3.3v. So the pot needs to be supplied with both GND and 3.3V in order to be able to do that. It will be a matter of working out where it is no longer supplying one or more of those signals

StanMOI, 2019-04-17:
Oh my god!
Thank you Kite!
I am so stupid...Shame on me...
As I changed from digital to anologic volume module, i just forgot to put wires between PCB and the volume module (AGND/AVOL/3V3) !
Shame on me :)
Know its works great!
Sorry to be so stupid :)

kiteretro, 2019-04-18:
:D very happy to hear!

closed 2019-04-18 by kiteretro