kiteretro/Circuit-Sword #90: Hard crashed after installing a theme & now I have no wifi

tlxxxsracer, 2019-07-10:
I was in the process of installing a few themes when the system froze and wouldnt accept input via keyboard or turning power switch off. Pulled battery. When it booted up I no longer had wifi

Log files. 

[syslog.txt](https://github.com/kiteretro/Circuit-Sword/files/3375734/syslog.txt)
[messages.txt](https://github.com/kiteretro/Circuit-Sword/files/3375735/messages.txt)

tlxxxsracer, 2019-07-10:
reflashed image. But every time I reboot my system wifi isnt connected and never shows wifi networks. I have to use `sudo rfkill unblock all` for wifi to work, but then the wifi icon doesnt even show up in the status bar but i do get an IP

kiteretro, 2019-07-10:
Hi, just a tip: if it crashed you can slide the switch OFF and then press the MODE button. This will do an instant power off/kill. This will save you having to pull the battery out and will achieve the same desired effect.

With wifi working, hold the MODE button and then press A (so that it turns to ON, you may have to hold down A for like 1 sec for it to register).. wait 10 seconds, and then repeat with B (to turn it off). Wait another 10 seconds and turn it ON again .. after doing this (enable, disable, enable) does it behave correctly? (icon + ip) ?

tlxxxsracer, 2019-07-14:
Thanks. Seems to fix it. Wasn't aware of those capabilities