kiteretro/Circuit-Sword #91: On off switch does not work

georgepapachristos, 2019-07-16:
hello,
the on off switch (safe shutdown) does not work and it never did Mode button works perfectly.
i dont know what to do as i have to leave the gameboy without screws in order to plug the battery in and out if i want to power it on and off. what should i do? anyone with the same problem? 
(the switch does nothing)

kiteretro, 2019-07-17:
Hi, I replied via email but I suspect my @kitesitemshop.com email address doesn't have the correct sending certificates.. anyway here is what I replied with:

If you unplug the FPC cable (that links to the back board) and then plug the battery/USB in, does it still do this? For example step #2 here: https://github.com/kiteretro/Circuit-Sword/wiki/How-to-set-up-the-Circuit-Sword

If not, then can you plug the FPC cable in, but unplug the button board (that 3pin black/green/blue cable). Does it still do it?

If it is fine up to there, then it sounds like a faulty switch unit and I'll send you out a replacement, just let me know the outcome of above and either we debug more or I send the replacement part :)

georgepapachristos, 2019-07-19:
Hi,
After I unplugged  the FPC cable and then plugged in the battery , nothing happend unless  i was pressing the power button, on the back board (from step 2) and if  only i keep pressing it, the gameboy booted and retropie was on the screen ready to play..
Now after i plugged the FPC cable back in and unplugged the 3pin black green blue cable wont start also unless i keep pressing the power button on backboard
what should I do next?

kiteretro, 2019-07-20:
Ok so the problem is specific to that little power switch board ONLY. Everything else is good :) If you carefully look at the switch board, in between the metal of the switch and the PCB may be some stray solder linking the metal of the switch to the middle pin? Is this something that you can see and fix? It's specifically something touching the middle pins of the switch.. I can send you a replacement, please send an email to kite@kiteretro.com and i'll def get that one and include your address and i'll send it out

georgepapachristos, 2019-07-23:
Dear KitePlease send me a replacement. I tried to desolder and i have even break the white thing  which you plug in the cable with the other white one. I totally destroyed. I m sorry. I ve sent you an email with address and name. Please send me a paypal link in order to cover its value and expenses. Sorry to disturb and thanks in advanceGeorge Papachristos

Στάλθηκε από το Ταχυδρομείο Yahoo σε Android

kiteretro, 2019-07-24:
I have received your email and will post out soon. No payment required at all, it was defective when it arrived therefore I fully cover it! :)

georgepapachristos, 2019-07-30:
thanks Kite , I m looking forward and feeling obligated...

georgepapachristos, 2019-08-07:
I have received the power switch board and it works perfectly.  
Thanks for everything....

closed 2019-08-07 by georgepapachristos