kiteretro/Circuit-Sword #96: Circuit Sword Lite no sound

FyLapS, 2019-10-27:
[image]

Hello,

I'm finishing up my GBZ build and I'm having a bit of trouble with sound. I do not get any sound out of speaker or through the headphone jack.

The mute toggle is on the screen but I cannot seem to turn it off.

When I run tester mode if get a fail on the sound check. Haven't been able to figure it out since the original test I did when I got the board.

I tried measuring the voltage when the speaker was unplugged and plugged in and each time which is consistently: 0.234 V from GND to S+ and 0.239 from GND to S-

I've looked through a few issues posted but cannot seem to find a solution. Any help would be greatly appreciated! 

Cheers

kiteretro, 2019-10-27:
Audio fail means it isn’t being detected on the USB bus. I would double check the soldering of the 3pin wire to the Pi. Ideally use a multimeter to check if any of the wires are shorted together! The most common thing is a bit of solder touching the exposed metal near the test pads, so check that? Maybe take a really close up pic of that soldering?

FyLapS, 2019-10-27:
[image]

Tried with my multimeter and there is no short between any of the wires, or even the yellow wire and the hole right next to it (on the right).

There are no issues with using the USB port that I can see btw (dont know if that helps). I managed to upload games via a USB stick.

Cheers

FyLapS, 2019-10-27:
I just noticed C20 and C21 short (the bottom connection). Should they be shorting there?

kiteretro, 2019-10-28:
C20 and C21 are fine like that, they're meant to be connected. Oh I see the issue! You have turned the "EXT-USB-EN" switch ON, you need to turn that OFF ;)

FyLapS, 2019-10-28:
All set thanks! Tester mode passed after i flipped the switch. Then it was a matter of turning WIFI off to get the sound to work!

Much appreciated and thanks for the support!!

kiteretro, 2019-10-28:
No problem, anything else just email :)

closed 2019-10-28 by kiteretro