kiteretro/Circuit-Sword #98: Overscan problem on 640x480 display

Joe2824, 2019-12-14:
If I change the overscan left or right the display starts to flicker. 
Overscan top and bottom have no issues.

I'm using the newest board with the new 640x480 display.

bin101, 2019-12-19:
I am facing the same issue but at my side I get the flickering also if I set top/bottom overscan.

kiteretro, 2019-12-19:
Hi why are you trying to use overscan? Does anything get cut off in normal use?

bin101, 2019-12-21:
For my part I wanted to cut off some pixel on the bottom because my glass lens printing is covering some space of the display. But now I think overscan is the wrong approach for this.

kiteretro, 2019-12-29:
Overscan isn't very good for this need because it will compress the image making the whole thing blurry, it doesn't reduce the visible screen area (e.g. if you set overscan bottom as 20px, it will still output 640x480 into a 'window' at 640x460, BUT the overscan doesn't scale quite like that, it usually ends up doing a linear scale so it will narrow the sides too..). You may have better luck editing the HDMI settings line, which is the long one that you have to uncomment to get the LCD to work, and adjust the '480' value lower, that MIGHT work but it will likely not.. the best solution is to make the visible area visible.. or you can adjust these things per app better in the SELECT + Y menu thing and set the scaling and offset per app instead

Joe2824, 2020-01-18:
The tipp with the HDMI settings does not work.
The boot looks not nice because the text is hidden from display glass.
Also the battery and wifi icons are hidden.
Is it possible to change the location of these icons? 

[image]

kiteretro, 2020-01-20:
Currently there isn't a way, i'll have a look to see if I can find some settings that may help

kiteretro, 2020-01-27:
Closing as this will be fixed as part of #67

closed 2020-01-27 by kiteretro