kiteretro/Circuit-Sword #102: joystick changes volume

njco90, 2020-04-12:
i updated through retropie and lost wifi. also my joystick keeps changing my volume levels down to 3 and 1 . i formatted and started fresh, wifi seems to back but the joystick is still messing w the volume. how do i fix this issue? thank you in advance!

njco90, 2020-04-13:
formatted fresh

njco90, 2020-04-13:
joystick is not working. already tried calibrating and enabling.

kiteretro, 2020-04-13:
Hi, could you follow the 'reprogram microcontroller' steps to update the arduino: https://github.com/kiteretro/Circuit-Sword/wiki/Circuit-Sword-FAQ#how-do-i-reprogram-the-microcontroller

and then run the `cs-configure.py` program and show a screenshot of the output (showing the information/state)

njco90, 2020-04-18:
I was able to calibrate and then set up in retroarch. Thank you for your
help