kiteretro/Circuit-Sword #103: Wifi Disabled After Update

DrRobotron, 2020-07-22:
So, instead of leaving well-enough alone I updated my GBCM3 tonight (kernal included), and now the WiFi isn't working. I guess I'm not the first one to run into this issue; how do I go about fixing it? 

The WiFi icon is crossed out, and when I try to add my home network again, I get the following error: "No wlan0 interface detected"

I recreated my wpa_supplicant.conf file without any luck. 

Thank you in advance for the help fixing my mistake.

kiteretro, 2020-07-22:
Hi,

Unfortunately the easiest way is to re-image to the latest image in this repo, and when you do the update make sure to press NO to the kernel one. You can update everything else so long as the kernel remains!

The get temporary connection back you should be able to use a USB wifi adapter that is supported by the standard kernel (most normal ones are)

DrRobotron, 2020-07-22:
Alright. Are there step-by-step instructions on how to do this somewhere? I'll use a dongle in the meantime, but I'd like to get it working again.

kiteretro, 2020-07-22:
See step #11 here: https://github.com/kiteretro/Circuit-Sword/wiki/How-to-set-up-the-Circuit-Sword You don't need to do the notepad++ changing, just write the image as detailed in that step. Note this will wipe everything to completely defaults! So backup anything that you want to keep

DrRobotron, 2020-07-22:
Gotcha. Will I be able to boot into retropie once the SD card is re-imaged, or will I be left with just linux. I can manage with re-tweaking things from within retroarch, but if I'm left with a completely blank slate otherwise, I may struggle. 

Should the LCD, keyboard, volume wheel, etc still work?

kiteretro, 2020-07-22:
Yes everything will work as before, it's what you would have had to do to get to the point you are at right now so it'll be familiar once you have done it. If in doubt of anything then do it to another SD card :)

DrRobotron, 2020-07-23:
Thank you sir! It took some work, but I'm back up and running. It wasn't as bad as I though it would be. The biggest pain was recreating all my .cfg and .txt files.

closed 2020-07-23 by DrRobotron