kiteretro/Circuit-Sword #104: Kernel/Software upgrade issue analysis

666dsa666, 2021-03-29:
Sorry I do not have the knowledge to have a better look at this, but I found that everything accessing SDIO/GPIO will break wifi on any updated img.

I went and imaged a clean sd, set everything up using a custom install.sh (network needed so I used a 3b+) and while it looks like everything is working, as soon as DPI-Cloner, HUD, SDIO or GPIO (for safe shutdown) are turned on then wifi will stop working.

I am not sure where the issue is, I've been trying to downgrade WiringPi but that doesn't seems to be the issue actually.

I've made a fork of this repo and tested everything I could to no avail beyond this point.

I believe resolving this issue would allow easier setup/upgrade.
I understand many people have no issues using the last image and I know I could too, but I had some time and figured it could be helpful.

kiteretro, 2021-03-29:
Hello, when you say "I went and imaged a clean sd" you mean that you imaged it with an image NOT found on this github page? If so, then the reason that wifi doesn't work is because you are missing the drivers. The RTL8723BS wifi chip drivers are NOT available as default and must be compiled in.

Take a look on this page: https://github.com/kiteretro/Circuit-Sword/tree/master/build

When I build the image, I get the "clean image" and then "upgrade the kernel" of that image with a kernel that I have built (which comes from here https://github.com/kiteretro/rpi-precompiled-kernels). This then gives me a "base image" that has the right kernel WITH wifi drivers. I then use that image with the `build-image.sh` script which ends up producing the final released build you find here on github.

If you want to do the same, all you need to do to any new/clean OS is to rebuild the kernel with the RTL8723BS module enabled.. from my repo linked above (once you have the kernel checked out):

```# <enable rl8723bs>
# make menuconfig
# Device Drivers -> Staging drivers -> Realtek RTL8723BS SDIO..
sed -i 's/# CONFIG_RTL8723BS is not set/CONFIG_RTL8723BS=m/' .config```

Will enable the module. It's not exactly an easy process, hence why I don't make further releases unless there is a NEED to do it.

666dsa666, 2021-03-29:
Thanks for the quick reply,

I thought I did build the kernel with the script but looks like I may have forgotten to copy it over and have to do it all again and check.

Do rtl8723bs* files need to be copied using the install.sh script ? It looks like files with the exact same filename are already present.

kiteretro, 2021-03-29:
the files are probably identical, if you already have them in place then don't worry about copying them. I cached them in this repo to make it easier to deploy, they didn't really need to be they are probably in my kernel files already..

mousemat86, 2021-04-08:
should this method work for latest retropie buster? 

If not, any plans to update? Thanks