kiteretro/Circuit-Sword #106: Screen going white

oki-dingo, 2021-07-23:
I’m having a screen issue with my freshly built 2021 circuit sword kit and the standard screen. After a couple minutes of the GB being on the screen starts to progressively become more and more white. If I turn it off and on again the screen will still be white. If I turn it off for 5-10 minutes the screen is a little bit white but immediately progresses to a less opaque white. If i turn it off for a while 30+ minutes the screen looks great but the cycle continues after a couple minutes and starts to turn white. 

Occasionally the screen will flicker for a few seconds. When it flickers it looks normal. I tried cleaning the board with a toothbrush and 100% isopropyl but it didn’t help.

The is a quick video showing the screen after letting it sit for a couple minutes and rebooting: [Here](https://youtube.com/shorts/Aqk3tuUolGA?feature=share) 

I am also having an issue with the “HD” screen. It only worked for a brief time and after rebooting it no longer displays an image. The back light comes on but the screen just stays black.

Kite, I already sent you an email, but wasn't sure if I’m supposed to post issues here first for tracking purposes.

mousemat86, 2021-09-01:
I have a similar probem – did you find a solution?

oki-dingo, 2021-09-01:
> I have a similar probem – did you find a solution?

After working with Kite and changing a couple parts (new CS-board, screen) it turned out to be the Raspberry Pi CM3 lite itself. I got the new CM3 in from Kite last week and it has been working perfectly for me so far.

mousemat86, 2021-09-02:
thanks for the reply. interesting, I'll try another CM3 and see if it's the same. 

previously it's been working with no issue

oki-dingo, 2021-09-02:
> thanks for the reply. interesting, I'll try another CM3 and see if it's the same.
> 
> previously it's been working with no issue

Mine old one worked perfectly for about 2 days then started and was constantly doing it.

mousemat86, 2021-09-09:
turns out the screen is faulty – time for a 640 x 480

oki-dingo, 2022-02-18:
Update: it came back!!! So after using it for about 5 hours max the screen fading to white came back again. Everything was changed except the battery and power switch. Might try reflashing the SD card