kiteretro/Circuit-Sword #108: Status LEDs do not come up

Zgv163, 2021-12-11:
Trying to set up the board in accordance with the [guide](https://github.com/kiteretro/Circuit-Sword/wiki/How-to-set-up-the-Circuit-Sword). However, at step 2 the LED does not go on, nether the LCD at later stages.

[image]

kiteretro, 2021-12-17:
Hi, can you try continuing with the guide untill it gets you to plug in the back board and power switch board? If you use the power switch (rather than the button) does it succeed in that step? IF so, then the 'button' itself is faulty and you can continue but replace the 'power button' steps for 'power switch' instead :)

FlashyFrank, 2023-03-26:
Hello, I am having the same issue. When I get to step 2 none of the LEDs come on when I push the button. After trying to follow the other steps I plug the LCD in and the sd card and nothing comes up on the screen. The screen comes on but it is just black. I put the sd card into my other circuit sword and it boots up to the test mode screen. Is there a work around or something I can do?