kiteretro/Circuit-Sword #109: Volume Wheel Not Controlling Volume

wolffman007, 2022-01-13:
Everything seems to be working fine within the image with the exception of the volume wheel controlling the volume.  As of now, volume wheel has no effect at all.

Volume can still be turned up/down within the Retropie menu.  Looks like it is set to "PCM" as the audio device with the OMX player set to "Both".  Headphones do not auto-switch but if I turn them on in the retropie menu it works.

Any fixes for getting the volume wheel to control?

kiteretro, 2022-01-14:
Have you followed the steps to enable the volume wheel? https://github.com/kiteretro/Circuit-Sword/wiki/Circuit-Sword-FAQ#how-do-i-enable-the-volume-wheel

wolffman007, 2022-01-14:
I did not!  Ran the script and its working perfectly - somehow missed that in the wiki.  Thanks for the reply and what an amazing board you created... really great stuff.

closed 2022-01-14 by wolffman007