kiteretro/Circuit-Sword #110: Make the sources of cs-hud and dpi-cloner available

jecaro, 2022-07-17:
I'm concerned about the future-proofness of this distribution for our beloved GBZ build.

It seems that the maintenance of it has been taken over by @weese in [his fork](weese/Circuit-Sword). Many thanks to him :pray: 

However I'm not aware of any place where we can find the sources for [cs-hud](https://github.com/kiteretro/Circuit-Sword/tree/master/cs-hud) and [dpi-cloner](https://github.com/kiteretro/Circuit-Sword/tree/master/dpi-cloner). Only binaries are present in this repo, those might not be compatible with future versions of retropie and not having the sources makes it impossible to add feature (I'm thinking about #67 for example).

So please consider publishing the sources of these two software that would help keeping that great device alive !

kiteretro, 2022-09-06:
Hi, yes I will do this, I'll update here when I have pushed the source, I appreciate your patience! :)

salami738, 2023-10-01:
Hi, is there an update on this?

kiteretro, 2023-10-14:
Better late than never! I have added a basic README describing how the src works. I have no doubt I may have forgotten to add some steps, for example libraries installed. I suspect you HAVE to run it on a Pi because it will need the gfx libs because it directly accesses the GPIO, but when building I imagine it will become clear pretty quickly what is needed: https://github.com/kiteretro/Circuit-Sword/tree/master/cs-hud/src#readme

weese, 2023-10-14:
That's awesome! Thanks a lot. With your hud sources, I can make the https://github.com/weese/Circuit-Sword distro future-proof.

jecaro, 2025-05-14:
For the record, I've been recently working on [a nixos config for the Circuit-Sword](https://github.com/jecaro/circuix-sword). I have integrated `cs-hud` into it. Unfortunately, OSD is not possible anymore as the API `cs-hud` uses to draw on top of the screen is now deprecated. But most stuff works fine 👌 (screen, sound, wifi) with a recent version of the software (linux and retroarch mostly).